espressif/esptool-js #5: Uncaught (in promise) TypeError: Cannot read property 'BOOTLOADER_FLASH_OFFSET' of null

seeya, 2021-08-03:
After selecting a `.bin` file to flash and clicking `Program`, the console display this error.

```
Uncaught (in promise) TypeError: Cannot read property 'BOOTLOADER_FLASH_OFFSET' of null
```

Any ideas how to fix this?

adwait-esp, 2021-08-05:
@seeya is this with the latest deployed version that you observed this issue?

seeya, 2021-08-05:
@adwait-esp yes, I cloned the latest version and tested again it's the same. It seems like the board didn't connect properly causing the variable `this.chip` to be null.

adwait-esp, 2021-08-05:
@seeya , are you able to perform other operations - like erase flash or get to the console with the same attached chip?

seeya, 2021-08-05:
@adwait-esp yes, using the flasher tool from Espressif Systems (https://www.espressif.com/en/support/download/other-tools), I'm able to flash the `firmware.bin` file at `0x1000`. 

Also with PlatformIO there's no issue.

torntrousers, 2021-08-07:
Same issue for me. Going to the live demo code at: https://espressif.github.io/esptool-js/. Plugin an ESP32, connect, select a firmware binary, and click program and nothing happens and in the Chrome developer tools console I see:

ESPLoader.js:1288 Uncaught (in promise) TypeError: Cannot read property 'BOOTLOADER_FLASH_OFFSET' of null
    at ESPLoader._update_image_flash_params (ESPLoader.js:1288)
    at ESPLoader.write_flash (ESPLoader.js:1358)
    at HTMLInputElement.programButton.onclick (index.js:279)

torntrousers, 2021-08-07:
Its a timing thing. If i put a breakpoint [here](https://github.com/espressif/esptool-js/blob/main/ESPLoader.js#L864) and just click continue when it stops there then it doesn't get that "Cannot read property" error any more. Still doesn't flash the ESP32 though, when I click program nothing happens and no errors in the Javascript console.

spiceywasabi, 2021-09-07:
Not sure if anyone still needs help on this error, but Web USB requires SSL so if you're running the local test server without SSL it will fail with this error message.

mikevanis, 2022-03-22:
Same issue here, on latest Chrome

a2800276, 2022-05-10:
What is the state of this project in general? Is `main` a working branch, or should we be using another one? I don't see how it even CAN work, e.g. connecting causes an infinite loop [here][inf]
There's no way to ever break out of this loop unless there's a timeout:
``` javascript
while (1) {
            try {
                const res = await this.transport.read({timeout: 1000});
                i += res.length;
                //console.log("Len = " + res.length);
                //var str = new TextDecoder().decode(res);
                //this.log(str);
            } catch (e) {
                if (e === "timeout") {
                    break;
                }
            }
            await this._sleep(50);
        }
```

unless I'm missing something entirely obvious/subtle, the loop above _should_ read:

```
while (1) {
            // loop until we read something from the device successfully
            try {
                const res = await this.transport.read({timeout: 1000});
                break; // successful read
            } catch (e) {
                if (e === "timeout") {
                    // caught and (expected) error, will continue to attempt connecting.
                    continue;
                }
                // caught a real error
                throw e
            }
            await this._sleep(50);
        }
```
[inf]:https://github.com/espressif/esptool-js/blob/081864a2ddad87a436137354323c8ee26fc92f42/ESPLoader.js#L817

adwait-esp, 2022-05-11:
This project is being maintained and the branch "gh_pages" which serves the https://espressif.github.io/esptool-js/ has the latest updated code.
We have observed intermittent connection issues for some revisions of ESP32 chipsets and we are working to resolve the same. Can you please let us know which ESP32 chipset you are trying to connect to?

a2800276, 2022-05-11:
It seems to me that the tool might be able to use some polishing, especially concerning handling of error cases and reporting on progress.

My main intention of writing here is to check whether it would make sense to contribute here, or if there are already people working on these (tbh rather obvious) deficits.

The boards I'm using are about as plain vanilla as it gets, e.g. in front of me is a WROOM Devkit, with a Silicon Labs CP210x Bridge:

```
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```

but also on a variety of known good devices and different bridges (at least FTDI),  all of which work fine even at high baud rates with the same cable and `esptool.py`

Out of curiosity, which chipsets/devkits are you using that work?

The connect/sync works _sometimes_ if run at the lowest baud rate. Even when it works, it takes a long time and provides no feedback during connection / sync. And not indication when an error occured. Typically, it's necessary to press Connect twice, because the connect procedure is stuck in the loop described above.

Even in the cases where the connect works, the upload fails. Again with no error indication.

balloob, 2022-07-19:
Above issues have been addressed. It now correctly raises if an unsupported chip is passed in (`chip == null`) and with #33 the latest UI issues has been fixed. Give it a try and please open new issues if you find some.

closed 2023-01-12 by igrr