espressif/esptool-js #38: Connect ESP32-S2  Error: Timeout

466034505, 2022-07-23:
esptool.js v0.1-dev
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting.........._
Detecting chip type... ESP32-S2
Chip is ESP32-S2
Features: Wi-Fi
Crystal is 40MHz
MAC: 68:67:25:2d:f4:0c
Uploading stub...
Error: Timeout

Wind-stormger, 2022-08-09:
When try to do it again:
```
esptool.js v0.1-dev
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting....Error: Failed to set control signals.
```

danielminson, 2023-02-27:
I am having this same issue with my board!
[image]
Is there anything I am doing wrong?

jonasniesner, 2023-04-28:
I am having exactly the same error, is there an update on this? In my case, it decides to randomly work one in 10 times.

musicmindmachine, 2023-06-23:
Same here.  Any stub or baud rate eventually times-out sometime during `flash_write`.  This is my first time using this, and frankly the JS/TS docs are a little vague. I've also tried in both production and dev webpack environments, with the same error.

Same, Is there anything I'm doing wrong?

Example Invoking Update Function: 
```typescript
export const updateDFUDeviceTest = async (
  devType = DFUDeviceTypes.SynergyV0,
  erase = false,
  newFirmware?: File,
  offset?: number
) => {
  console.log("Updating Device FW!");
  // const romBaudRate = 115200;

  //Attempt to get the most recent firmware
  if (devType.getUpdatedFirmware) {
    const updates = await devType.getUpdatedFirmware();
    devType = { ...devType, ...updates };
    if (devType.firmwareFile)
      console.log(`Updated Firmware is ${devType.firmwareFile.name}`);
  }

  try {
    // Look for devices in DFU mode
    const port = await navigator.serial.requestPort({
      filters: webSerialDFUDeviceFilters,
    });
    if (!port) throw new Error("No DFU Device");

    const loader = new ESPLoader(
      new Transport(port),
      devType.programBaud,
      undefined,
      devType.romBaud
    );
    loader.MAX_TIMEOUT = 2000;

    await loader.connect(devType.connectResetMode, 10, false);
    // Get the Chip's STA MAC ADDRESS (SERIAL NUMBER)
    const name = loader.chip.CHIP_NAME;
    const sizes = Object.keys(loader.chip.FLASH_SIZES).join(",");

    console.log(`Connected to ${name} with Flash Sizes: ${sizes}`);
    const desc = await loader.chip.get_chip_description(loader);
    if (desc !== "ESP32-S2FH32") throw new Error("Not a Synergy ESP32-S2");
    console.log(`Devie Description matches Synergy: ${desc}`);

    const freq = await loader.chip.get_crystal_freq(loader);
    console.log(`CPU Cryastal Freq: ${freq}Mhz`);

    const features = (await loader.chip.get_chip_features(loader)).join(",");
    console.log(`Chip Features: ${features}`);

    const mac = await loader.chip.read_mac(loader);
    console.log(`Chip STA MAC Address (USB S/N): ${mac}`);

    // Load default bin files into our command
    const flashSections = [] as { address: number; data: string }[];
    if (devType.bootloaderFile) {
      const bBA = await devType.bootloaderFile.arrayBuffer();
      const bBytesString = new TextDecoder().decode(bBA);
      console.log(
        `Using Bootloader Image ${devType.bootloaderFile.name} at 0x${
          devType.bootloaderOffset?.toString(16) || 0
        }`
      );
      flashSections.push({
        address: devType.bootloaderOffset || 0x1000,
        data: bBytesString,
[... 75 lines cut ...]

Consistently produces the result:
```
Downloading FW from /fw/synergy/synergy_fw_0_95_64.bin
VM76369 SynergyV0.ts:35 Downloading Bootloader from /fw/synergy/synergy-bootloader-tinyuf2.bin
VM76369 SynergyV0.ts:42 Downloading Partition Table from /fw/synergy/synergy-partitions.bin
VM76369 SynergyV0.ts:49 Downloading UF2 Updater App from /fw/synergy/synergy-tinyuf2-updater.bin
VM76362 DeviceSerialUpdate.ts:128 Updated Firmware is synergy_fw_0_95_64.bin
esploader.js:129 esptool.js
esploader.js:129 Serial port WebSerial VendorID 0x303a ProductID 0x2
esploader.js:129 Connecting...
2esploader.js:129 .
esploader.js:129 

VM76362 DeviceSerialUpdate.ts:142 Connected to ESP32-S2 with Flash Sizes: 1MB,2MB,4MB,8MB,16MB
VM76362 DeviceSerialUpdate.ts:145 Devie Description matches Synergy: ESP32-S2FH32
VM76362 DeviceSerialUpdate.ts:147 CPU Cryastal Freq: 40Mhz
VM76362 DeviceSerialUpdate.ts:149 Chip Features: Wi-Fi,Embedded 4MB Flash,ADC and temperature sensor calibration in BLK2 of efuse
VM76362 DeviceSerialUpdate.ts:151 Chip STA MAC Address (USB S/N): 84:f7:03:d2:94:cc
VM76362 DeviceSerialUpdate.ts:198 Using Firmware Image synergy_fw_0_95_64.bin at 0x10000
esploader.js:129 Compressed 719084 bytes to 351416...
esploader.js:129 Took 0.0s to erase flash block
esploader.js:129 Writing at 0x10000... (4%)
react_devtools_backend_compact.js:2367 Error: Timeout
```

Jason2866, 2023-06-23:
You are doing nothing wrong. Flashing S2 via CDC does not work.

Jason2866, 2023-07-16:
Is there activity and if yes when can a fix be expected?

igrr, 2023-07-16:
@Jason2866 At the moment, no one from Espressif is working on this issue.

The main reason why this type of issues doesn't get resolved quickly is that they require someone with deep knowledge of both ESP boot and with web development skills (JS/TS) to contribute. Since these two topics are pretty far apart, people with both of these skills are hard to come by.

I think after https://github.com/espressif/esptool-js/pull/107 it will get easier to compare the traces against esptool.py implementation, so issues like this will get easier to troubleshoot. Then the task will become somewhat simpler. Instead of "figure out why this timeout is happening" it will be something like "change this code so that it produces the same trace as that reference implementation".

Longer term, the plan is to get the whole flashing algorithm from https://github.com/esp-rs/espflash (compiled to WebAssembly) and hide it behind the same-ish API as we have now. This way we will have to troubleshoot fewer independent implementations of the flashing protocol, which means fewer issues or faster response times.

Jason2866, 2023-07-16:
@igrr  Thank you for this infos. Knowing the reasons why "something" makes no visible process does help ;-)

brianignacio5, 2023-12-05:
Hello @Jason2866 @musicmindmachine @danielminson @newpsgo @Wind-stormger @jonasniesner 

Have you guys tried the latest master to see if this issue is fixed ? There is a now a tracing feature to track the communication too. Please check the live example.

tomaszduda23, 2024-04-25:
I have similar issue

```
esptool.js
esploader.js:158 Serial port WebSerial VendorID 0x303a ProductID 0x2
esploader.js:158 Connecting...
esploader.js:158 Debug: _connect_attempt default_reset false
esploader.js:158 Debug: Timeout
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 .
esploader.js:158 Debug: _connect_attempt default_reset true
esploader.js:158 Debug: Timeout
esploader.js:158 Debug: Sync
esploader.js:158 Debug: Sync err Error: Timeout
esploader.js:158 _
esploader.js:158 Debug: Sync
esploader.js:158 Debug: 538052359
esploader.js:158 

esploader.js:158 Debug: Chip Magic 7c6
esploader.js:158 Detecting chip type... 
esploader.js:158 ESP32-S2
esploader.js:158 Chip is ESP32-S2FH32
esploader.js:158 Features: Wi-Fi,Embedded 4MB Flash,ADC and temperature sensor calibration in BLK2 of efuse
esploader.js:158 Crystal is 40MHz
esploader.js:158 MAC: 84:f7:03:f4:f1:fc
esploader.js:158 Uploading stub...
esploader.js:158 Debug: mem_begin 4336 1 6144 40028000
esploader.js:158 Debug: check_command enter RAM download mode
esploader.js:158 Debug: check_command write to target RAM
install-web-dialog.ts:192 Error: Timeout
    at Zi.read (webserial.js:128:27)
    at async $i.read_packet (esploader.js:221:19)
    at async $i.check_command (esploader.js:391:15)
    at async $i.mem_block (esploader.js:412:40)
    at async $i.run_stub (esploader.js:667:46)
    at async $i.main_fn (esploader.js:728:43)
    at async u._handleInstall (install-web-dialog.ts:188:7)
_handleInstall @ install-web-dialog.ts:192
await in _handleInstall (async)
firstUpdated @ install-web-dialog.ts:165
_$AE @ state.js:3
performUpdate @ state.js:3
scheduleUpdate @ state.js:3
_$E_ @ state.js:3
await in _$E_ (async)
requestUpdate @ state.js:3
o @ state.js:3
Ht @ state.js:3
Ut @ base.js:2
u @ install-web-dialog.ts:29
T @ index.ts:39
await in T (async)
_handleRetry @ install-web-dialog.ts:174
handleEvent @ animate.js:1
install-web-dialog.ts:246 Closing port
```

sasodoma, 2024-05-12:
@brianignacio5 After cloning and building the main branch, and running the TS example, I can confirm that I still get a timeout when trying to connect to an ESP32-S2.
trace output of esptool.js: https://pastebin.com/tq6Pcx48
trace output of esptool.py: https://pastebin.com/hr0n3qKW

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

sasodoma, 2024-11-15:
Using the original esptool.js at https://espressif.github.io/esptool-js/ I get the following output after simply connecting:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting....
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH32
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:ea:b5:8a
Uploading stub...
Error: Timeout
```

Using your fork at https://brianignacio5.github.io/esptool-js/ I get this output:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH32
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:ea:b5:8a
Uploading stub...
Error: No serial data received.
```

Finally, using python esptool with the `flash_id` command I get:
```
esptool.py v4.7.0
Serial port COM8
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:ea:b5:8a
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)
WARNING: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
```

brianignacio5, 2024-11-20:
Hi @sasodoma thank you for the feedback. I've continued to do more changes and released them in my fork.

Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

sasodoma, 2024-11-20:
Sure thing, here is the debug log from your fork after I connect to an ESP32-S2 Mini:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic 7c6
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:ea:b8:3e
Uploading stub...
Debug: mem_begin 4444 1 6144 40028000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Error: No serial data received.
```
And here is the output from the developer console.
[brianignacio5.github.io-1732132850538.log](https://github.com/user-attachments/files/17836168/brianignacio5.github.io-1732132850538.log)

brianignacio5, 2024-11-21:
I think I may have found the culprit. Esp32s2 was missing the post connect function that update the ESP_RAM_BLOCK for S2 chips.

Changes are up in https://brianignacio5.github.io/esptool-js/ Could you please try again ? Apologies for the trouble.

sasodoma, 2024-11-21:
I did a quick check before going to work and it seems to have worked. Initial connection worked fine, when flashing I got a checksum mismatch but I think it still booted. I will do more detailed checks in the evening when I get back home.

sasodoma, 2024-11-23:
Sorry for the delay, I was a bit busy. I have now tried to do some more flashing and I get a checksum mismatch every time. I believe that the data is not being written at all, because the program on the ESP32-S2 doesn't change or get corrupted. And when I flash the same firmware first via Python esptool, and then via your fork I get a checksum match and the tool says the flashing succeeded, but I believe that it simply didn't change anything. Here is the output when I try to flash new firmware:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic 7c6
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v0.0)
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:ea:b8:3e
Debug: In _post_connect using USB OTG ?true
Uploading stub...
Debug: mem_begin 4444 3 2048 40028000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Debug: check_command write to target RAM
Debug: check_command write to target RAM
Debug: mem_begin 160 1 2048 3ffe2bfc
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Running stub...
Debug: check_command leave RAM download mode
Stub running...
Changing baudrate to 921600
Changed
Debug: EspLoader program
Debug: Data Length 14720
Debug: Image Length 14720
Debug: _update_image_flash_params keep undefined undefined
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Debug: Image MD5 0627a801b7177d2ce3ec293dfad9ebc3
Compressed 14720 bytes to 10221...
Debug: check_command enter compressed flash mode
Took 0.0s to erase flash block
Debug: Write loop 4096 0 1
Writing at 0x1000... (100%)
Debug: flash_defl_block 78 da
Debug: check_command write compressed data to flash after seq 0
Wrote 14720 bytes (10221 compressed) at 0x1000 in 0.111 seconds.
Debug: check_command calculate md5sum
File  md5: 0627a801b7177d2ce3ec293dfad9ebc3
Flash md5: f5ed8965716d408600c273a61d31651c
Error: MD5 of file does not match data in flash!
```

brianignacio5, 2024-11-25:
Hi @sasodoma I've found that the IS_STUB was not being set so the flash values were not being picked correctly. Now I fixed it.

Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

brianignacio5, 2024-11-26:
Hi @Jason2866 @tomaszduda23 @musicmindmachine @danielminson @jonasniesner 
Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

Jason2866, 2024-11-26:
Flashing S2/S3 via CDC "modem" does work now in my tests!
It would be great when the RTC WDT reset method would be added to bring the devices out of flash mode. RTC WDT is implemented not long ago in esptool.py with this commit https://github.com/espressif/esptool/commit/8298cdcc68a8063df1ca932d42e21555e15269b3

closed 2024-11-28 by brianignacio5

Elamarang1, 2025-04-21:
esp32 upload was wokring but one time upload EN button click then only running currently update code other vice not relating next another time i upload than i get this issues .

Uncaught runtime errors:
×
ERROR
Invalid head of packet (0x65): Possible serial noise or corruption.
    at Transport.read (http://localhost:3000/static/js/bundle.js:11074:19)
    at async ESPLoader.readPacket (http://localhost:3000/static/js/bundle.js:9420:11)
    at async ESPLoader.checkCommand (http://localhost:3000/static/js/bundle.js:9675:18)
    at async ESPLoader.flashDeflBegin (http://localhost:3000/static/js/bundle.js:9824:5)
    at async ESPLoader.writeFlash (http://localhost:3000/static/js/bundle.js:10247:18)
    at async handleProgram (http://localhost:3000/static/js/bundle.js:1550:5)

please any one tell me how to solve that issue?

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]