espressif/esptool-js #39: Chip type not detected

ithinkido, 2022-07-25:
Testing on EDGE Version 102.0.1245.44, with both an ESP8266 and ESP32 board with both local and the [hosted](https://espressif.github.io/esptool-js/) version - The chip type is not detected and returns a value of null.
[image]

balloob, 2022-07-26:
You must be using a cached version. connectButton.onClick is defined on line 78, not [line 90](https://github.com/espressif/esptool-js/blob/803d4e3f3f8f644cc3cb79de0eb8d639cd4b48d7/index.js#L90).

ithinkido, 2022-07-26:
The error is pointing to line 98 and not line 90. 
[image]
This is in a new private window with no cache

ithinkido, 2022-07-26:
This issue remind me of a similar one I faced once due to Flatpak installed apps being sand boxed. Turns out it was the same. I have reinstalled EDGE using the .deb package and it is now working. This is quite deceptive as the "connect to port" window does still seem to work, and correctly  lists all ports and shows the /dev/ttyUSB0 port but fails later. Sorry for the false alert.

closed 2022-07-26 by ithinkido