espressif/esptool-js #40: Can't upload .bin to 0x00010000 ESP32

EarlOld, 2022-07-26:
When I use ArduinoIDE, all works correctly: 

```
Compressed 940304 bytes to 602887...

Writing at 0x00010000... (2 %)
Writing at 0x000197fd... (5 %)
Writing at 0x0002591d... (8 %)
Writing at 0x0002ef7e... (10 %)
Writing at 0x0003e65a... (13 %)
Writing at 0x0004402c... (16 %)
Writing at 0x0004972b... (18 %)
Writing at 0x0004f616... (21 %)
Writing at 0x00055243... (24 %)
Writing at 0x0005a931... (27 %)
Writing at 0x0005ffed... (29 %)
Writing at 0x000652b4... (32 %)
Writing at 0x0006a626... (35 %)
Writing at 0x0006f83d... (37 %)
Writing at 0x00074b1b... (40 %)
Writing at 0x00079ebf... (43 %)
Writing at 0x0007f100... (45 %)
Writing at 0x00084429... (48 %)
Writing at 0x000894c8... (51 %)
Writing at 0x0008e874... (54 %)
Writing at 0x0009465e... (56 %)
Writing at 0x00099b05... (59 %)
Writing at 0x0009ed9f... (62 %)
Writing at 0x000a42b7... (64 %)
Writing at 0x000a94b2... (67 %)
Writing at 0x000ae91e... (70 %)
Writing at 0x000b3e98... (72 %)
Writing at 0x000b948f... (75 %)
Writing at 0x000bec93... (78 %)
Writing at 0x000c4a5f... (81 %)
Writing at 0x000ca397... (83 %)
Writing at 0x000d298c... (86 %)
Writing at 0x000db638... (89 %)
Writing at 0x000e095b... (91 %)
Writing at 0x000e625b... (94 %)
Writing at 0x000eb97b... (97 %)
Writing at 0x000f0e03... (100 %)
```

When I use esptool-js: 
```
Compressed 940308 bytes to 602890...
Writing at 0x10000... (2%)
Writing at 0x14000... (5%)
Writing at 0x18000... (8%)
Writing at 0x1c000... (10%)
Writing at 0x20000... (13%)
Writing at 0x24000... (16%)
Writing at 0x28000... (18%)
Writing at 0x2c000... (21%)
Writing at 0x30000... (24%)
Writing at 0x34000... (27%)
Writing at 0x38000... (29%)
Writing at 0x3c000... (32%)
Writing at 0x40000... (35%)
Writing at 0x44000... (37%)
Writing at 0x48000... (40%)
Writing at 0x4c000... (43%)
Writing at 0x50000... (45%)
Writing at 0x54000... (48%)
Writing at 0x58000... (51%)
Writing at 0x5c000... (54%)
Writing at 0x60000... (56%)
Writing at 0x64000... (59%)
Writing at 0x68000... (62%)
Writing at 0x6c000... (64%)
Writing at 0x70000... (67%)
Writing at 0x74000... (70%)
Writing at 0x78000... (72%)
Writing at 0x7c000... (75%)
Writing at 0x80000... (78%)
Writing at 0x84000... (81%)
Writing at 0x88000... (83%)
Writing at 0x8c000... (86%)
Writing at 0x90000... (89%)
Writing at 0x94000... (91%)
Writing at 0x98000... (94%)
Writing at 0x9c000... (97%)
Writing at 0xa0000... (100%)
```

In **.py** writing starts from 0x00010000 in **.js** - 0x10000. 

Looks like something wrong with int to address convert.
[image]

[image]

Please advice, thanks.

TD-er, 2022-07-26:
> In **.py** writing starts from 0x00010000 in **.js** - 0x10000.

I don't see the problem here.
`0x10000` == `0x00010000` == `0x000000010000`

So what is the problem you're seeing?

EarlOld, 2022-07-26:
I found a bug in my code. Thanks.

closed 2022-07-26 by EarlOld