espressif/esptool-js #41: Connect error with ESP32-C3 via CDC: Timeout

Jason2866, 2022-08-11:
```
esptool.js v0.1-dev
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting.........._______......._______......._______......._______......._______......._______......._______Error: Failed to connect with the device
```
Screenshot from Chrome Console
[image]

@balloob fyi

Jason2866, 2022-08-11:
When using Tasmota WebInstaller which uses `esp-web-flasher` under the hood flashing does work.
[image]

balloob, 2022-08-11:
Maybe the C3 stub is incorrect and so gets stuck once written to RAM https://github.com/espressif/esptool-js/blob/main/targets/esp32c3.js

brentru, 2022-09-14:
Having the same issue with ESP32-C3.

liux-pro, 2022-12-01:
same issue

TD-er, 2022-12-02:
Related? https://github.com/espressif/esptool-js/pull/54

Jason2866, 2022-12-02:
@brianignacio5  Can you implement in your PR?

igrr, 2023-01-12:
This should be now fixed by #54, thanks @liux-pro!

igrr, 2023-01-12:
I thought I tested the final merge commit on both C3 with USB_SERIAL_JTAG and CP2102, but looks like I didn't.

Updating the stub binary seems to solve the issue. I'll do more checks with different boards and open a PR.

igrr, 2023-01-12:
@liux-pro may I ask you to try https://github.com/espressif/esptool-js/pull/65 and see if that fixes the issue on your end? I've only updated the stub binary for ESP32-C3 there.

liux-pro, 2023-01-12:
> @liux-pro may I ask you to try #65 and see if that fixes the issue on your end? I've only updated the stub binary for ESP32-C3 there.

Fully functional !

Jason2866, 2023-01-13:
Confirmed working. Good for release when debugging Infos are removed.

igrr, 2023-01-13:
#65 has been merged, hopefully this is fixed for good!

closed 2023-01-13 by igrr

Histywy, 2025-04-23:
esptool.je

Serial port Webserial VendorID 0x10c4 ProductID Oxea60

Connecting... Error: Failed to execute 'open' on 'SerialPort': The port is already open.

I'm having this problem and don't know what the solution is