espressif/esptool-js #42: Add WiFi Provisioning

franz-ms-muc, 2022-08-15:
Add WiFi Provisioning

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html

Would make a lot of sense.
After flashing use Serial Console Protocomm to scan and Connect.

franz-ms-muc, 2022-11-03:
as an Example we can look to [https://github.com/esphome/esp-web-tools](https://github.com/esphome/esp-web-tools)

[https://esphome.github.io/esp-web-tools/](https://esphome.github.io/esp-web-tools/)

works quite well, but their Provisioning Standard ist somehow weird and non ESP-IDF like.

BrucePerens, 2022-12-31:
Since there is a console serial terminal, why not run a WiFi configuration wizard on the console? I know it's not as pretty, but it's simpler and works across all tools that provide a serial console.

I implemented a client for the Improv WiFi protocol used by esphome. It's lacking in robustness because it doesn't have any means to do retries if a protocol exchange is missed and it can't handle duplicate or gratuitous packets - and the first packet is often missed because the ESP32 sends it before Webserial is ready. It also uses a rather simplistic packet structure rather than something extensible like JSON.

I thought about redesigning Improv WiFi, but ended up implementing the WiFi wizard on the ESP32 console.

franz-ms-muc, 2023-01-04:
> Since there is a console serial terminal, why not run a WiFi configuration wizard on the console? I know it's not as pretty, but it's simpler and works across all tools that provide a serial console.

i do exactly this in the Moment. ESP Provisioning by QR-Code and BLE.

igrr, 2023-02-28:
I am thinking of moving this issue into https://github.com/espressif/esp-launchpad, since it looks like this will be at least partially application-specific. Some applications might support CLI based provisioning, others might handle it via NVS. Let's keep esptool-js repo with just the "flashing" functionality.