espressif/esptool-js #44: Missing version in package.json

dumbasPL, 2022-09-22:


balloob, 2022-09-23:
You should be able to add it based on a commit [like this](https://github.com/esphome/esp-web-tools/blob/main/package.json#L34)

dumbasPL, 2022-09-23:
> You should be able to add it based on a commit [like this](https://github.com/esphome/esp-web-tools/blob/main/package.json#L34)

well yes and no. This does work with npm, but yarn will refuse to install it

```
$ yarn init -y
yarn init v1.22.19
warning The yes flag has been set. This will automatically answer yes to all questions, which may have security implications.
success Saved package.json
Done in 0.02s.
$ yarn add github:espressif/esptool-js#0c1b972a05d691c85da23fcc937d91dcf7e283eb
yarn add v1.22.19
info No lockfile found.
[1/4] Resolving packages...
error Can't add "esptool-js": invalid package version undefined.
info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.
```
and the only reason npm works is because it skips all checks for git repositories
```
$ npm i github:espressif/esptool-js#0c1b972a05d691c85da23fcc937d91dcf7e283eb
npm WARN skipping integrity check for git dependency ssh://git@github.com/espressif/esptool-js.git

added 3 packages, and audited 4 packages in 1s
```

The npm documentation specifies the `name` and `version` as required fields so the package.json is invalid either way.
```
A package.json file must contain "name" and "version" fields.

The "name" field contains your package's name, and must be lowercase and one word, and may contain hyphens and underscores.

The "version" field must be in the form x.x.x and follow the [semantic versioning guidelines](https://docs.npmjs.com/about-semantic-versioning).
```
[(source)](https://docs.npmjs.com/creating-a-package-json-file)

balloob, 2022-09-23:
Want to open a PR and add version 0.1 to it?

CC @adwait-esp

dumbasPL, 2022-09-23:
> Want to open a PR and add version 0.1 to it?

done

PR #46 (2022-11-07): Add package version
closes #44

reference: https://docs.npmjs.com/creating-a-package-json-file
closed 2022-11-07 by adwait-esp