espressif/esptool-js #45: Upload a binary from Arduino IDE?

torntrousers, 2022-09-22:
Should this work to upload a binary from the Arduino IDE using the ESP32 / Arduino framework? It doesn't for me.

I have this sketch:
```
void setup() {
  Serial.begin(115200);
  Serial.println("\nsetup");
}

void loop() {
  Serial.println("loop");
  delay(1000);
}
```
I use menu option "Sketch -> Export compiled Binary" to show the .bin file, which I've attached here.

Use the Live demo link and connect to my local board, choose the binary file, and click program. It all seems to flash ok, screen shot:

[image]

but then the board seems to be borked and wont reset and nothing comes up in a serial monitor or terminal.

balloob, 2022-09-23:
ESP32 binaries contain usually of multiple files. If you exported a single file, it should be flashed to 0x0000. ESP Web Tools uses ESPtool.js like that https://esphome.github.io/esp-web-tools/

torntrousers, 2022-09-23:
Thanks @balloob . I tried using 0x0000 but no change. It did prompt me to enable verbose mode in the Arduino IDE and look at its flashing log, which is this:
```
Sketch uses 249953 bytes (3%) of program storage space. Maximum is 6553600 bytes.
Global variables use 16456 bytes (0%) of dynamic memory, leaving 4505528 bytes for local variables. Maximum is 4521984 bytes.
esptool.py v3.3
Serial port COM3
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:f8:7e:ec
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0004dfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 336.6 kbit/s)...
Hash of data verified.
Flash params set to 0x024f
Compressed 17120 bytes to 11841...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11841 compressed) at 0x00001000 in 0.8 seconds (effective 174.8 kbit/s)...
Hash of data verified.
Compressed 250336 bytes to 134723...
Writing at 0x00010000... (11 %)
Writing at 0x0001dce3... (22 %)
Writing at 0x00024e8f... (33 %)
Writing at 0x0002a451... (44 %)
Writing at 0x0002f8b9... (55 %)
Writing at 0x00038471... (66 %)
Writing at 0x00040d05... (77 %)
Writing at 0x0004661c... (88 %)
Writing at 0x0004bd4b... (100 %)
Wrote 250336 bytes (134723 compressed) at 0x00010000 in 3.1 seconds (effective 637.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 197.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

So the main sketch looks like its loaded at  0x00010000. I tried that and that does work. Yay.

Looking in more detail, the Arduino IDE flashes four areas:
```
Wrote 8192 bytes (47 compressed) at 0x0000e000
Wrote 17120 bytes (11841 compressed) at 0x00001000 
Wrote 250336 bytes (134723 compressed) at 0x00010000
Wrote 3072 bytes (129 compressed) at 0x00008000
```
Any tips on where to find those other three files it flashes?

TD-er, 2022-09-23:
Depends on the settings for your flash (e.g. DIO/QIO mode and flash frequency)
If you always have the same settings and boards (!!), you could also flash one unit and read back the flash from that unit and then use the first `0x00010000` bytes of that download to make your own "bootloader".

The reason why this might be a good idea is that esptool.py may have altered the flash settings on the fly while flashing, based on the actually detected flash chip.

The reason why this might be a bad idea is that not all boards are equal.
Also it may cause issues when you switch from the older SDK to the latest based on IDF 4.4.x

No idea where ArduinoIDE stores these files, but you can have a look at the script I use to make a single binary for ESP32 when using PlatformIO:

https://github.com/letscontrolit/ESPEasy/blob/3678488f6a67ac9b899fab3d0ccd176c4437b1b4/tools/pio/post_esp32.py#L12-L17

torntrousers, 2022-09-25:
Thanks @TD-er , thats helpful.

igrr, 2023-01-12:
Cross-linking the feature request to export a single binary from Arduino IDE: https://github.com/espressif/arduino-esp32/issues/6481.

I think the question has been answered as far as possible, I'll close this issue.

closed 2023-01-12 by igrr