espressif/esptool-js #47: Flash error with ESP32-S3: Timeout

walchko, 2022-09-29:
Appears to be the same with S2 and C3 ... there is a timeout error when trying to flash the chip.

adetauriac, 2022-10-17:
Hello

I ve the same error :-( no idea to fix it

nicpottier, 2022-10-24:
Same problem here using this board:
  https://www.lilygo.cc/products/t-display-s3

igrr, 2023-01-12:
> Serial port WebSerial VendorID 0x303a ProductID 0x1001

Looks like you are flashing it via USB_SERIAL_JTAG. This should work after #54 and #65.

igrr, 2023-01-25:
Both related PRs have been merged, I think this should be fixed now. Please reopen if the issue still exists.

closed 2023-01-25 by igrr

makermelissa, 2023-01-26:
It looks like this is still an issue. I am trying to flash an ESP32-S3 and still getting a timeout error:
[image]

I'm trying to flash the combined.bin inside of https://github.com/adafruit/tinyuf2/releases/download/0.12.0/tinyuf2-adafruit_feather_esp32s3_tft-0.12.0.zip at address 0, which works fine on https://adafruit.github.io/Adafruit_WebSerial_ESPTool/. However, I'd love to retire that tool in favor of this one once it is reliable enough.

makermelissa, 2023-01-26:
The board I'm testing with is https://www.adafruit.com/product/5483. However, there's a number of bootloaders for espboards at https://github.com/adafruit/tinyuf2/releases if you want to test with a similar one.

makermelissa, 2023-01-26:
#83 looks kind of promising for fixing the issue though.

igrr, 2023-01-27:
@makermelissa I have tried the PR #83 with the binary linked tinyuf2 binary and flashing was successful. I've pushed this branch also to https://igrr.github.io/esptool-js/ for you to try.

makermelissa, 2023-01-27:
Ok, I just tried it and it worked great. Thank you. It looks like this tool is getting very close now.

MichaelDu9226, 2023-03-02:
@igrr The timeout problem still seems to exist on ESP32 S3 USB JTAG/DEBUG mode.
I have tested https://igrr.github.io/esptool-js/ and https://espressif.github.io/esptool-js/.
[image]

[image]

curlyz, 2023-03-16:
> @igrr The timeout problem still seems to exist on ESP32 S3 USB JTAG/DEBUG mode. I have tested https://igrr.github.io/esptool-js/ and https://espressif.github.io/esptool-js/. [image]
> 
> [image]

Confirm this issue is also existing on C3, esptool-js keep timeout although python version work, but with --no-stub flag onl