espressif/esptool-js #48: ESP32-C3: "app image at 0x10000 has invalid magic byte" if the app image flashed at first

LeoYan, 2022-10-08:
### Module
ESP32-C3-WROOM-02

### Issue

- Use esptool.py to flash ESP32-C3, it works after the module is reseted.
- Use https://espressif.github.io/esptool-js/ to flash same images with the address the order:(0x0, 0x8000, 0xe000, 0x10000),  it works after the module is reseted.
- Use https://espressif.github.io/esptool-js/ to flash same images with the address the order:(0x0, **0x10000**, 0x8000, 0xe000 ),  it cannot works after the module is reseted.  
- the error is "esp_image: image at 0x10000 has invalid magic byte"

### Esptool-js Failed Log

- Flash(Adress: 0x0,0x10000, 0x8000, 0xe000 ) log in the Console:
```
esptool.js v0.1-dev
Serial port WebSerial VendorID 0x10c4 ProductID 0xea60
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:bc:37:ac
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Compressed 19844 bytes to 12011...
Writing at 0x0... (100%)
Wrote 19844 bytes (12011 compressed) at 0x0 in 0.46 seconds.
Hash of data verified.
Compressed 583300 bytes to 321867...
Writing at 0x10000... (5%)
Writing at 0x14000... (10%)
Writing at 0x18000... (15%)
Writing at 0x1c000... (20%)
Writing at 0x20000... (25%)
Writing at 0x24000... (30%)
Writing at 0x28000... (35%)
Writing at 0x2c000... (40%)
Writing at 0x30000... (45%)
Writing at 0x34000... (50%)
Writing at 0x38000... (55%)
Writing at 0x3c000... (60%)
Writing at 0x40000... (65%)
Writing at 0x44000... (70%)
Writing at 0x48000... (75%)
Writing at 0x4c000... (80%)
Writing at 0x50000... (85%)
Writing at 0x54000... (90%)
Writing at 0x58000... (95%)
Writing at 0x5c000... (100%)
Wrote 583300 bytes (321867 compressed) at 0x10000 in 9.144 seconds.
Hash of data verified.
Compressed 3076 bytes to 128...
Writing at 0x8000... (100%)
Wrote 3076 bytes (128 compressed) at 0x8000 in 0.059 seconds.
Hash of data verified.
Compressed 8196 bytes to 30...
Writing at 0xe000... (100%)
Wrote 8196 bytes (30 compressed) at 0xe000 in 0.145 seconds.
Hash of data verified.
Leaving...
```

- The output after the module restarted:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d1546
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b8
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d40
entry 0x403ce000
I (35) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (35) boot: compile time 15:17:18
I (35) boot: chip revision: 3
I (37) boot.esp32c3: SPI Speed      : 80MHz
I (42) boot.esp32c3: SPI Mode       : DIO
I (47) boot.esp32c3: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (75) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (82) boot:  2 app0             OTA app          00 10 00010000 00140000
I (90) boot:  3 app1             OTA app          00 11 00150000 00140000
I (97) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (105) boot: End of partition table
I (109) boot: No factory image, trying OTA 0
E (114) esp_image: image at 0x10000 has invalid magic byte (nothing flashed here?)
E (122) boot: OTA app partition slot 0 is not bootable
E (128) esp_image: image at 0x150000 has invalid magic byte (nothing flashed here?)
E (136) boot: OTA app partition slot 1 is not bootable
E (142) boot: No bootable app partitions in the partition table
```

### Esptool-js Success Log

- Flash(Adress: 0x0, 0x8000, 0xe000, 0x10000) log in the Console:

```
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:bc:37:ac
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Compressed 19844 bytes to 12011...
Writing at 0x0... (100%)
Wrote 19844 bytes (12011 compressed) at 0x0 in 0.449 seconds.
Hash of data verified.
Compressed 3076 bytes to 128...
Writing at 0x8000... (100%)
Wrote 3076 bytes (128 compressed) at 0x8000 in 0.056 seconds.
Hash of data verified.
Compressed 8196 bytes to 30...
Writing at 0xe000... (100%)
Wrote 8196 bytes (30 compressed) at 0xe000 in 0.145 seconds.
Hash of data verified.
Compressed 583300 bytes to 321867...
Writing at 0x10000... (5%)
Writing at 0x14000... (10%)
Writing at 0x18000... (15%)
Writing at 0x1c000... (20%)
Writing at 0x20000... (25%)
Writing at 0x24000... (30%)
Writing at 0x28000... (35%)
Writing at 0x2c000... (40%)
Writing at 0x30000... (45%)
Writing at 0x34000... (50%)
Writing at 0x38000... (55%)
Writing at 0x3c000... (60%)
Writing at 0x40000... (65%)
Writing at 0x44000... (70%)
Writing at 0x48000... (75%)
Writing at 0x4c000... (80%)
Writing at 0x50000... (85%)
Writing at 0x54000... (90%)
Writing at 0x58000... (95%)
Writing at 0x5c000... (100%)
Wrote 583300 bytes (321867 compressed) at 0x10000 in 9.012 seconds.
Hash of data verified.
```

### Esptool.py Log

- Flash info:

```
esptool.py esp32c3 -p /dev/cu.SLAB_USBtoUART -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 bbScale.bin 0x8000 partition_table/partition-table.bin 0xe000 ota_data_initial.bin
esptool.py v3.2-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:bc:37:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x0009efff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 19840 bytes to 12006...
Writing at 0x00000000... (100 %)
Wrote 19840 bytes (12006 compressed) at 0x00000000 in 0.6 seconds (effective 268.6 kbit/s)...
Hash of data verified.
Compressed 583296 bytes to 321864...
Writing at 0x00010000... (5 %)
Writing at 0x0001a238... (10 %)
Writing at 0x00024900... (15 %)
Writing at 0x0002adb0... (20 %)
Writing at 0x00031669... (25 %)
Writing at 0x00038c5e... (30 %)
Writing at 0x0003ff69... (35 %)
Writing at 0x00046f36... (40 %)
Writing at 0x0004e094... (45 %)
Writing at 0x0005502e... (50 %)
Writing at 0x0005bf9e... (55 %)
Writing at 0x00062ac3... (60 %)
Writing at 0x00069d9e... (65 %)
Writing at 0x000718f7... (70 %)
Writing at 0x000781de... (75 %)
Writing at 0x0007e8f5... (80 %)
Writing at 0x000851f0... (85 %)
Writing at 0x0008b180... (90 %)
Writing at 0x00092b02... (95 %)
Writing at 0x00099d34... (100 %)
Wrote 583296 bytes (321864 compressed) at 0x00010000 in 9.5 seconds (effective 490.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 401.4 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0000e000 in 0.1 seconds (effective 565.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
```

- The output after the module restarted:

```
mac$ idf.py monitor
Executing action: monitor
Serial port /dev/cu.usbserial-1420
/dev/cu.usbserial-1420 failed to connect: [Errno 16] could not open port /dev/cu.usbserial-1420: [Errno 16] Resource busy: '/dev/cu.usbserial-1420'
Serial port /dev/cu.SLAB_USBtoUART
Connecting.........
Detecting chip type... ESP32-C3

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b8
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d40
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (30) boot: compile time 15:17:18
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (70) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 app0             OTA app          00 10 00010000 00140000
I (85) boot:  3 app1             OTA app          00 11 00150000 00140000
I (93) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (100) boot: End of partition table
```

LeoYan, 2022-11-03:
The reason:

1.
if fileArray[i].data.length % 4 == 0,  there will be extra 4 bytes, so overlap with the following image
`image = fileArray[i].data + '\xff\xff\xff\xff'.substring(0, 4 - fileArray[i].data.length % 4);`

btw:
2.  the address of C3 bootloader is 0x0 rather then 0x1000

adwait-esp, 2022-11-10:
Latest fixes have been pushed to gh-pages branch. Can you please confirm if this issue is fixed?

closed 2023-01-12 by igrr