espressif/esptool-js #49: Time out when trying to flash spiffs.bin for an ESP32

SeByDocKy, 2022-10-20:
Hello, I am trying to flash the spiffs part of a project with this tool.
From pio in verbose mode, the command line executed is : 
```--chip esp32 --port "“COM18”" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio                  --flash_size detect 2686976 .pio/build/routeur_solaire/spiffs.bin`````

Seems the offset adress is 2686976 = 0x290000.

When trying with esptool-js got a timeout error...
Maybe I am wrong somewhere, but is-it a bug ?

vidsmet2, 2022-11-21:
@SeByDocKy 

I've had the same issue. Changing the baudrate to 115200 solved this.

SeByDocKy, 2022-11-23:
> @SeByDocKy
> 
> I've had the same issue. Changing the baudrate to 115200 solved this.

Thanks I will try asap :)

bpstermometria, 2022-11-24:
I have the same issue. any solution?

igrr, 2023-01-12:
This might be solved by https://github.com/espressif/esptool-js/pull/57. I'd like to compare how esptool.py solves the same issue, though, before merging that PR.

closed 2023-02-09 by igrr