espressif/esptool-js #58: Bug: connection error after typescript rewrite

domagojk, 2022-12-22:
Hi there! I wanted to try this tool and after I tested it online it worked correctly:

```
esptool.js v0.1-dev
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 40:91:51:a5:d8:14
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 115200
Changed
```

but when I tried it locally, I got this:

```
esptool.js v0.1-dev
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting..._connect_attempt default_reset false

Sync
Sync err Error: Timeout

.Sync
Sync err Error: Timeout

.Sync
Sync err Error: Timeout

.Sync
Sync err Error: Timeout

.Sync
Sync err Error: Timeout

.Sync
Sync err Error: Timeout

.Sync
Sync err Error: Timeout

._connect_attempt default_reset true

Sync
Sync err Error: Timeout

_Sync
Sync err Error: Timeout

_Sync
Sync err Error: Timeout
```

So I tried checking out to a commit which is on `gh-pages` (the one before  [first attempt to re write in typescript](https://github.com/espressif/esptool-js/commit/24e047581faa6d1bd86686bcbc5d956856610ccf)) and it worked again!

It looks like there may have been a bug introduced in the recent commit 😬  Unfortunately, I am not familiar enough with the codebase to determine the cause. So, I wanted to bring this to your attention in case it helps with troubleshooting. 

Still, this looks promising, keep up the good work! 🙂

Jason2866, 2022-12-22:
The issue is known https://github.com/espressif/esptool-js/pull/55#issuecomment-1339877083

domagojk, 2022-12-22:
Oh, I see, ok then 👍