espressif/esptool-js #59: Error when trying to connect to esp32 device (Sync Errror:Timeout)

falaah151, 2023-01-10:
Errors in console:

  index.js:112 Error: Timeout
      at ei.read (bundle.js:2:65884)
      at async ti.command (bundle.js:2:49127)
      at async ti.read_reg (bundle.js:2:49475)
      at async ti.connect (bundle.js:2:51399)
      at async ti.detect_chip (bundle.js:2:52198)
      at async ti.main_fn (bundle.js:2:60077)
      at async connectButton.onclick (index.js:107:12)

Errors in output:

  esptool.js v0.1-dev
  Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
  Connecting..._connect_attempt default_reset false
  
  Sync
  Sync err Error: Timeout
  
  .Sync
  538052359
  
  Error: Timeout

balloob, 2023-01-12:
Fixed in #62

igrr, 2023-01-12:
The fix is now merged!

closed 2023-01-12 by igrr