espressif/esptool-js #64: New NPM release with USB_SERIAL_JTAG support

igrr, 2023-01-12:
Now that https://github.com/espressif/esptool-js/pull/54 is merged, we might want to make a new NPM release.

brianignacio5, 2023-01-13:
We could create a tag with new version when major issues are fixed. CI Action [publish](https://github.com/espressif/esptool-js/blob/main/.github/workflows/publish.yml) should work for publish it and release an artifact in GitHub releases.

Jason2866, 2023-01-13:
+1 for a new release

igrr, 2023-01-17:
Both linked PRs are now merged, so I think we set the [version](https://github.com/espressif/esptool-js/blob/4bac4457dbed43fdd10b7d64ba3d320291838907/package.json#L3) to 0.2.0 and tag the new release.

brianignacio5, 2023-01-17:
There is also another small bug fix in #72 we should merge before the release.

igrr, 2023-01-25:
Released in https://www.npmjs.com/package/esptool-js/v/0.2.0

closed 2023-01-25 by igrr