espressif/esptool-js #80: Changing baud rate causes timeout

balloob, 2023-01-19:
We use esptool-js to power [ESP Web Tools](https://github.com/esphome/esp-web-tools). When it starts installing using esploader, it passes an existing port. The port has been opened, used to query [improv](https://www.improv-wifi.com/), and then closed.

`ESPLoader` will, as part of `main_fn()`, call `await this.change_baud();`. This command is causing the connection to timeout and make the connection unusable.

The fix for ESP Web Tools is to [apply a patch](https://github.com/esphome/esp-web-tools/blob/main/patches/0001-not-change-baud.patch) to the installed node module to comment out the changing of the baud rate. When done, everything works fine.

I've been looking into the code trying to understand what's going on:

 - ESPLoader takes `baudrate` and `rom_baudrate` param. Second one defaults to `115200`
 - `main_fn()` calls `detect_chip()`
 - `detect_chip() calls `connect()
 - `connect()` calls `transport.connect(this.rom_baudrate)`
 - `transport.connect()` opens the port at the requested baud rate and stores it as `transport.baudrate`
 - ESPLoader detects chip, runs stub.
 - ESPloader calls `change_baud`.
 - `change_baud` sends a packet. Then disconnects and opens again.

Question: would it be possible to skip changing the baud if `this.baudrate == this.rom_baudrate` ?

igrr, 2023-01-19:
> Question: would it be possible to skip changing the baud if this.baudrate == this.rom_baudrate ?

Sounds reasonable, I can't think of any downside.

(The whole connection flow could use some clean-up, by the way. It's odd that we have both detect_chip and connect, but we need to call an aptly named main_fn.)

marko-savic, 2023-01-19:
Same problem here, and fix by @balloob suggestion.

Jason2866, 2023-01-19:
The change does fix our noticed issue when using current esptool.js
With the change it does work well. A new release would be nice.

PR #81 (2023-01-25): Only change baudrate if it's different between ROM and STUB
Fixes #80
closed 2023-01-25 by igrr