espressif/esptool-js #86: It takes 3 minutes to decide that a connection cannot be established

balloob, 2023-02-03:
ESPLoader is initialized by calling `main_fn`. This function calls `detect_chip`, which in turn calls `connect` which triggers the connection sequence.

If a device requires the BOOT mode to be held down to start flashing, the connection will fail. Currently it takes 4-5 minutes on my machine before it raises the error `Failed to connect with the device`. This seems too long.

The default attempts number for `connect` is set to 7. This means that 7 times it tries the following:

 - Calls `connect_attempt` with `esp32r0_delay` set to `false`
 - Calls `connect_attempt` with `esp32r0_delay` set to `true`

`main_fn` allows passing in a `mode` that is passed down to `connect_attempt`. It is defined as type `string`, it defaults to `default_reset`. It can be set to `no_reset` to skip resetting the device inside `connect_attempt`. The resetting takes quite some time, with hardcoded sleeps. If `esp32r0_delay` is true, there is a call to sleep 2 seconds!

Because our connection runs `connect_attempt` twice for each attempt, we are trying to attempt to connect 14 times. That's already twice as many as the Python tool does by default.

Below a screenshot of the console logs. This is not a full single attempt (out of 7), as it cuts off the 2nd call to `connect_attempt` with `esp32r0_delay` set to `true`. The screenshot already spans 26 seconds. It has another 6 full attempts to go!

[image]

Looking at `esptool`, it also resets between every attempt, but it uses different reset strategies that leverage different delays https://github.com/espressif/esptool/blob/master/esptool/loader.py#L567. The strategies are defined at https://github.com/espressif/esptool/blob/master/esptool/reset.py#L61. By default the sleep delay between RTS setting is `0.05` and `0.55` for a 2nd reset attempt. There are also calls to 0.1 and 0.2s sleeps. There definitely is never a 2s sleep 👀 Also there is no reference to what `esp32r0_delay` is, it's not in the Python tool.

Summary:
- What is `esp32r0_delay`?
- Why are we doing two connection attempts for each attempt, doubling the number of attempts what esptool does?
- Are there ways we can speed this up further?

igrr, 2023-02-03:
> What is esp32r0_delay?

https://github.com/espressif/esptool/commit/2d72f56ce6b6103d908194f6d4237a2032ce9b89#diff-b035aa3701bf24cdb68151b147c0cfe693bbb1b27812a779f13d4255bdff1fc7L597

Agree that some delays here are probably excessive, we can speed this up.

balloob, 2023-02-04:
If esptool has removed it, can we remove it too? Or maybe we can guard it for Windows users only based on the user agent string ?

RushikeshPatange, 2023-08-21:
@balloob, @igrr, and @brianignacio5, could you please provide any updates regarding this particular issue?

brianignacio5, 2023-12-05:
Do we still have these issues ? @RushikeshPatange @balloob ?

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]
closed 2024-11-28 by brianignacio5