espressif/esptool-js #87: Problem with most current (11.2.0.167) CP210X driver on Windows 10

dkalliv, 2023-02-04:
On Windows 10 systems using the latest  (11.2.0.167) CP210X driver I can not connect to ESP32 devices with a Silicon Labs CP2102 or CP2104 USB to UART bridge. I receive `Error: Timeout`
PlatformIO or the Arduino IDE can connect and flash without issues. When downgrading the driver to version 6.7.4.261, esptool-js works as expected.

brentru, 2023-02-22:
Hi - also experiencing this with ESP8266 boards and the latest driver.

brentru, 2023-02-22:
@dkalliv How were you able to downgrade the driver? I don't see any links for version `6.7.4.261` on the SiLabs site.

dkalliv, 2023-02-22:
On one system I just did a downgrade (only possible if the driver was previously upgraded) for the others I downloaded the drivers from here:
[https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads](url)

danielminson, 2023-02-25:
I am also experiencing this error! To fix it I selected a driver already available in my computer on device manager
[image]
Like this:
[image]

igrr, 2023-07-13:
@RushikeshPatange no, I haven't looked at this, I don't have a Windows PC to test with at the moment.

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

closed 2024-11-28 by brianignacio5

tyeth, 2026-11-09:
Think this might still exist, my ESP32-S31 Korvo-1 experienced it @brianignacio5 (windows11 Chrome)
https://issues.chromium.org/issues/583357789 being the underlying issue (works fine in firefox)
I'm using latest (not wasm PR) with some minor changes https://github.com/tyeth/read-partitions-esptool-js/tree/uat/read-partitions-s31

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]