espressif/esptool-js #88: Upload of bin to esp32 DOIT DEVKIT doesn't work as expected

danielminson, 2023-02-17:
Hello, I am having trouble with the flashing process of specific board esp32 DOIT DEVKIT. Even with the upload process getting to 100%, the program flashed does not run even after resseting the board. The .bin file is attached as a zip and is a blink code (D2 turning on and off every second). I even tried flashing in different addresses 0x0 and 0x1000. Thank you for the attention.

[image]
[firmware.zip](https://github.com/espressif/esptool-js/files/10768236/firmware.zip)

danielminson, 2023-02-17:
I made it work with the writing address 0x00010000, but the process showed that the address was 0x1000 (as shown in the screeshot attached). I seem to understand that there's a problem with the setup, probably removing all the zeros. 
[image]

igrr, 2023-02-17:
Hi @danielminson,

> The .bin file is attached as a zip and is a blink code (D2 turning on and off every second). I even tried flashing in different addresses 0x0 and 0x1000.

Your binary is an _application_ binary. If you have built the binary using ESP-IDF or Arduino or PlatformIO, there are actually multiple binaries that need to be flashed:

* the _bootloader_ (at offset 0x1000)
* the _partition table_ (at offset 0x8000)
* the _application_ (the offset might be different, often it is 0x10000)

> I made it work with the writing address 0x00010000, but the process showed that the address was 0x1000 (as shown in the screeshot attached)

I'm sorry, where exactly in the screenshot do you see 0x1000 (i.e. 4096, decimal)? The screenshot attached to your 2nd comment shows offset 0x10000 (i.e. 65536, decimal), which is the same as 0x00010000.

[image]

igrr, 2023-02-17:
If the bootloader and the partition table are already flashed and you know that you haven't changed either the bootloader settings (like flash DIO/QIO mode) or the partition table, then you can just reflash the application.

danielminson, 2023-02-25:
Okay. Thank you very mych

closed 2023-02-25 by danielminson