espressif/esptool-js #91: Timeout uploading stub on ESP32-S2

makermelissa, 2023-03-03:
I was testing out a few boards. Most are working great, but I'm having an issue with it timing out when uploading the stub on the ESP32-S2. I tried multiple times and at 115200 and 921600.

This was tested with an Adafruit MagTag and the combined.bin file in https://github.com/adafruit/tinyuf2/releases for the board.

Here's the error:
[image]

Here's the Console:
[image]

Thanks.

ralphwetzel, 2023-03-09:
The same happens here with a M5Atom Lite:

```
esptool.js
Serial port WebSerial VendorID 0x403 ProductID 0x6001
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: Wi-Fi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:a9:9d:38
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 1500000
Changed
Not changing the image
Compressed 17440 bytes to 12131...

Error: Timeout
```

Used to run smoothly with an earlier version => 0.1 !

makermelissa, 2023-03-09:
@ralphwetzel I think that's a different error as you're getting to the `Stub running...` message and beyond. You may consider trying a slower baud rate, as 1500000 is quite fast and could be causing the timeouts.

Looking through the issues, I think mine may actually a duplicate of #38, which I seemed to have missed when I created this.

ralphwetzel, 2023-03-10:
> You may consider trying a slower baud rate, as 1500000 is quite fast and could be causing the timeouts.

I did - despite the device usually works perfectly with that baud rate. The error yet stays.
As said, all was fine with an earlier version...

jonasniesner, 2023-04-30:
I have the same error

Jason2866, 2023-06-01:
@igrr  Tested with your latest PR and the timeout error is still there

igrr, 2023-06-15:
This is probably the same issue as https://github.com/espressif/esptool-js/issues/38. I'll close this one as duplicate, please subscribe to that older issue to get notified of the progress.

closed 2023-06-15 by igrr

TCB13, 2024-01-10:
ESP32-S2 Mini board from Aliexpress [here](https://www.wemos.cc/en/latest/s2/s2_mini.html) failing in ESPHome. To get around this was to manually flash the initial that the web installer (https://web.esphome.io/?dashboard_wizard) normally installs.

I got the image from https://github.com/esphome/firmware/actions/runs/7443385432 then downloaded the esphome-web artifact and extracted the `esphome-web-esp32s2.bin` image to burn using the [Espressif Flash Download Tool](https://www.espressif.com/en/support/download/other-tools) with the address `0x0`.

[image]

Make sure DoNotChgBin is checked.

After this I was able to go connect my phone to it, set it to my home wifi network and then go into Home Assistant > ESPHome and flash the final firmware as required.