espressif/esptool-js #95: Support for Flash encryption and Secure Boot?

mrninhvn, 2023-05-26:
How to flash secure boot enabled chip and have flash encryption?

risoul, 2023-10-29:
I second that

mrninhvn, 2023-10-30:
> I do not want show original code of the .bin

I solved flashing, however the .bin file is always displayed when you check resources with dev tools.

risoul, 2023-10-31:
As i understand, you talk about .bin server side? then it is not encrypted?

I found this about encryption : https://www.motius.com/post/how-to-build-a-secure-iot-prototype-with-arduino-and-esp32
Using arduino to code then esp-idf to encrypt

But I do not know if it can be flashed with esptool-js

fukaminakrize, 2023-12-10:
Hello, I would too like to ask about any plans for supporting flash encryption in esptool-js.

mrninhvn, 2024-04-09:
My modification to support Secure Boot and Flash Encryption. Enjoy!
[https://github.com/mrninhvn/esptool-js/tree/encrypted-flash](url)

chipweinberger, 2024-04-25:
is anyone else getting "unsupported command error"?

https://github.com/espressif/esptool-js/issues/139

mrninhvn, 2024-04-25:
> @mrninhvn , how do I run your fork? Can you host it on github? That would be great.
> 

You can build this example or modify esp-web-tool to use the enc-esptool-js module.
I do a quick deloy here: [https://mrninhvn.github.io/esptool-js/](url)

brianignacio5, 2024-09-11:
Could you try https://github.com/espressif/esptool-js/pull/145 ? @mrninhvn @risoul  @chipweinberger