espressif/esptool-js #96: Flashing S3 via CDC does not work reliable

Jason2866, 2023-05-29:
Most of the time a timeout happens. See [issue](https://github.com/esphome/esp-web-tools/issues/331) in esp-web-tools where esptool.js is used.
There is an another unsolved [issue](https://github.com/espressif/arduino-esp32/issues/8237) in Arduino Github.

@igrr Can you help?

igrr, 2023-05-29:
@Jason2866 Do you have a binary which we can flash to reproduce this issue?

sblantipodi, 2023-05-29:
@igrr here it is...
https://github.com/sblantipodi/glow_worm_luciferin/tree/master/docs/static/firmware_build_beta

the factory ones are the correct ones to use.

igrr, 2023-05-29:
Interesting, I am not yet able to reproduce the issue with this binary... Do you meet the issue with a board which doesn't yet have anything flashed to it, or with a board which is flashed with some specific application?

From the logs in the linked issue (timeout well into the upload process) I think that this recent fix to the flasher stub might help: https://github.com/espressif/esptool/commit/80dc2104e163acddf6a6a6fa26c1e3f7068173c3.
I'll try some more to reproduce the issue and then see if updating the stubs makes any difference.

sblantipodi, 2023-05-29:
@igrr I have the same problem with a "clean board" and with a board pre-flashed with a CDC firmware.

igrr, 2023-05-29:
After flashing the board with [GlowWormLuciferinFULL_ESP32_S3_firmware-factory.bin](https://github.com/sblantipodi/glow_worm_luciferin/blob/master/docs/static/firmware_build_beta/GlowWormLuciferinFULL_ESP32_S3_firmware-factory.bin) and resetting the board, esptool.js can't put it into download mode:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x8167
Connecting.........._______......._______......._______......._______......._______......._______......._______Error: Failed to connect with the device
```
I have also tried esptool.py, which can't connect either. Opening the serial terminal and toggling DTR/RTS signals I also couldn't make the board reset into the download mode, so I assume the issue here is that the application doesn't handle DTR/RTS to reset into download mode. This looks like the topic of the other issue you just opened (https://github.com/espressif/esptool-js/issues/97) so I'm not going to address it here.

---

If I manually place the board into ROM download mode, it can be flashed. Also I haven't managed to reproduce the issue with  flashing a board when flash is erased. I have erased flash using `esptool.py` first, then flashed the same application. I have also tried flashing a large (4 MB) binary. On 4 attempts, it got to the end of the flashing process without an issue:
```
Writing at 0x3ffafe... (100%)
Wrote 4194304 bytes (4195590 compressed) at 0x0 in 26.028 seconds.
Hash of data verified.
Leaving...
```

I am thinking that perhaps I'm not able to reproduce the issue due to a somewhat different eFuse configuration. Could you please run `espefuse.py --port YOUR_PORT summary` and attach the output you get as a file?

sblantipodi, 2023-05-29:
if I put the board in CDC mode by pressing Boot, RST, leave RST, this is the output:

```
PS C:\dev\dpsoftware\ArduinoRaspberry\PUBLIC\glow_worm_luciferin> espefuse.exe --port //./COM11 summary
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Detecting chip type... ESP32-S3
espefuse.py v3.3.1

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms
DIS_LEGACY_SPI_BOOT (BLOCK0)                       Disables Legacy SPI boot mode                      = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)                        Selects the default UART for printing boot msg     = UART0 R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)                     Disables USB OTG download feature in UART download = False R/W (0b0)
                                                    boot mode
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot
BLOCK_USR_DATA (BLOCK3)                            User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                                    Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Identity fuses:
SECURE_VERSION (BLOCK0)                            Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                                   ure)
MAC (BLOCK1)                                       Factory MAC Address
   = f4:12:fa:a8:d5:b8 (OK) R/W
WAFER_VERSION (BLOCK1)                             WAFER version                                      = 1 R/W (0b001)
PKG_VERSION (BLOCK1)                               ??? Package version                                = ESP32-S3 R/W (0x0)
BLOCK1_VERSION (BLOCK1)                            ??? BLOCK1 efuse version                           = 1 R/W (0b001)
OPTIONAL_UNIQUE_ID (BLOCK2)                        ??? Optional unique 128-bit ID
   = 16 63 52 48 9e f1 42 d5 42 7b 76 cf b8 84 90 05 R/W
BLOCK2_VERSION (BLOCK2)                            ??? Version of BLOCK2                              = 5 R/W (0b101)
CUSTOM_MAC (BLOCK3)                                Custom MAC Address
   = 00:00:00:00:00:00 (OK) R/W

Security fuses:
SOFT_DIS_JTAG (BLOCK0)                             Software disables JTAG by programming odd number o = 0 R/W (0b000)
                                                   f 1 bit(s). JTAG can be re-enabled via HMAC periph
                                                   eral
HARD_DIS_JTAG (BLOCK0)                             Hardware disables JTAG permanently                 = False R/W (0b0)
DIS_DOWNLOAD_MANUAL_ENCRYPT (BLOCK0)               Disables flash encryption when in download boot mo = False R/W (0b0)
                                                   des
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables encryption and decryption, when an SPI boo = Disable R/W (0b000)
                                                   t mode is set. Enabled when 1 or 3 bits are set,di
                                                   sabled otherwise
SECURE_BOOT_KEY_REVOKE0 (BLOCK0)                   Revokes use of secure boot key digest 0            = False R/W (0b0)
SECURE_BOOT_KEY_REVOKE1 (BLOCK0)                   Revokes use of secure boot key digest 1            = False R/W (0b0)
[... cut ...]
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.
PS C:\dev\dpsoftware\ArduinoRaspberry\PUBLIC\glow_worm_luciferin>
```

igrr, 2023-05-29:
Thanks for trying this! The output does look pretty much the same as on the board I have here.

One other thing to try, could you flash the same binary on your end using https://igrr.github.io/esptool-js/?

I have just pushed a commit to that fork, updating esptool stubs to the latest v4.6 release which includes the fix for Super WDT triggering during the upload process.

Jason2866, 2023-05-29:
@igrr  on an erased S3 this one fails to flash via CDC

http://ota.tasmota.com/tasmota32/tasmota32s3cdc.bin

igrr, 2023-05-29:
@Jason2866 I'm not sure if it fails for you with https://igrr.github.io/esptool-js/, as suggested in the last comment, or with https://espressif.github.io/esptool-js/, or with another flasher? I have just tried flashing this binary to an erased S3 (`esptool.py --port /dev/tty.usbmodem4401 erase_flash`) using https://igrr.github.io/esptool-js/ and that worked fine:
```
Writing at 0x1660cd... (100%)
Wrote 1477376 bytes (1039405 compressed) at 0x0 in 12.194 seconds.
Hash of data verified.
Leaving...
```
Could you please clarify which tool it fails for you with, and post a log? (Was it a connection timeout, or failure when flashing was already underway?)

Jason2866, 2023-05-29:
@igrr esptool.py does work well. It fails when using https://github.com/esphome/esp-web-tools
I have removed the CDC builds currently from my [webflasher](https://jason2866.github.io/Tasmota-specials/)

igrr, 2023-05-29:
@Jason2866 Just to clarify my previous comment, I have successfully flashed the binary _using https://igrr.github.io/esptool-js/_, not esptool.py. I only mentioned esptool.py in relation to erasing the contents of flash prior to flashing with esptool-js.

Could you please try flashing http://ota.tasmota.com/tasmota32/tasmota32s3cdc.bin using https://igrr.github.io/esptool-js/ and https://espressif.github.io/esptool-js/ ?
If it fails on your end, please attach the log output you get. Thank you.

Jason2866, 2023-05-29:
@igrr  your https://igrr.github.io/esptool-js/ is working!! :-)

igrr, 2023-05-29:
What about https://espressif.github.io/esptool-js/? If it didn't work, could you please attach the log output?

Jason2866, 2023-05-29:
Well, it works too. But when used with https://github.com/esphome/esp-web-tools it fails :-(

sblantipodi, 2023-05-29:
@igrr I tried flashing my firmware with that tool, 
it flashes the firmware without problems but once finished the ESP goes in boot-loop.

I see the devices continously connecting and disconnecting from Windows... weird. if I flash the same file with the web installer I linked previously it works ok.

sblantipodi, 2023-05-29:
I tried flashing the tasmota firmware and same boot loop issues when using this tool:
 https://igrr.github.io/esptool-js/

Jason2866, 2023-05-29:
@sblantipodi You have to use the factory image and flash at address 0x0

sblantipodi, 2023-05-29:
oops, sorry, flashing at address 0x0 works with both firmware.

closed 2023-06-15 by igrr