espressif/esptool-js #105: Live Demo no longer working

QuirkyCort, 2023-07-05:
In commit c34c8c7, index.ts was changed to use .textContent when reading the offset and file data. Since both the offset and file elements are html input elements with no textContent, the demo no longer works. It should be using .value and .data instead.

Relevant lines:
https://github.com/espressif/esptool-js/blob/58df4cdb6f902c7b38dfaf787f212f9f8fb8028f/examples/typescript/src/index.ts#L279

https://github.com/espressif/esptool-js/blob/58df4cdb6f902c7b38dfaf787f212f9f8fb8028f/examples/typescript/src/index.ts#L290

RushikeshPatange, 2023-07-08:
Yes, when clicking on the **program button**, nothing is happening due to the issue mentioned by @QuirkyCort in the [live demo example.](https://espressif.github.io/esptool-js/) Could you please take a look at it, @brianignacio5? Thank you! 😄

brianignacio5, 2023-07-10:
Oh apologies for the issue! Will update asap !

closed 2023-07-17 by brianignacio5

sdrshnptl, 2023-11-12:
Live demo not working again.
Any thought why this might be broken?
[image]

PR #106 (2023-07-17): fix example offset file data
Fix #105 

This PR should fix the issues in the live demo related to offset and file data.