espressif/esptool-js #120: Using this tool to burn MicroPython firmware on ESP8266 always fails

drbdrb, 2023-11-29:
I use https://espressif.github.io/esptool-js/ to burn MicroPython firmware, version 
 [v1.21.0 (2023-10-05) .bin](http://www.micropython.org/resources/firmware /ESP8266_GENERIC-20231005-v1.21.0.bin),
 the programming can work normally, but the chip cannot run normally after burning. So far, I have also tried to use esptool.py to burn this firmware. Using esptool.py to burn the same firmware is It can run normally, but burning using esptool.js tool does not work. Hope the author can solve this problem?

brianignacio5, 2023-12-06:
Can you provide more detail what issue are you facing with esptool-js @drbdrb  ?

drbdrb, 2023-12-09:
It's like this, I use: https://espressif.github.io/esptool-js/ The tool always fails to burn the official firmware of micropython for ESP8266, and the chip cannot function properly after burning! However, using the esptool. py tool to burn the same official firmware for micropython can still run normally.

The burning addresses are all: 0x0000, and the firmware is the same. You can download from this address: https://www.micropython.org/resources/firmware/esp8266_generic-20231005-v1.21.0.bin

I found that when using the esptool. py tool, I need to set the ` -- flash_ Size=detect ` is set to: detect, but I changed the flash in the esptool js tool_ Size='detect 'is still unsuccessful.

The chip is ESP8266EX,

Burn write baud rate: 115200,

Location: 0x0000

I recorded a video and sent it to your @brianignacio5 email. Please check and help me see what caused this error?

drbdrb, 2024-01-20:
What did the author do recently? This project hasn't been updated for almost two months? Is there any hope of solving the problem I mentioned above?

Jason2866, 2024-01-29:
@drbdrb esptool.py `flash size detect` reads out the real flash size from the connected device and patches the firmware. This is not done by esptool.js This is as designed and described and not a bug.
You have to provide a firmware with the firmware size built which matches the device flash size.

ItzArty, 2025-03-11:
> @drbdrb esptool.py `flash size detect` reads out the real flash size from the connected device and patches the firmware. This is not done by esptool.js This is as designed and described and not a bug.
> You have to flash a firmware with the firmware size built which matches the device flash size.

A little late to the party, sorry! What you're saying is that I need to have a binary of the same file size as the flash size, or that I need to choose the specific firmware for the ESP provided by MicroPython?

Jason2866, 2025-03-11:
@ ItzArty The magic header of the firmware needs to reflect the flash size and mode.
Imho best is to build a merged binary with all needed files with esptool.py 
esptool.js is not capable to do this

ItzArty, 2025-03-12:
> @ItzArty The magic header of the firmware needs to reflect the flash size and mode.
> Imho best is to build a merged binary with all needed files with esptool.py 
> esptool.js is not capable to do this

Thank you for the reply. I would also like to note espwebtools seems to have a solution for this problem (if anyone else is looking into this).

Jason2866, 2025-03-12:
@ItzArty espwebtools uses esptool.js under the hood ;-) The solution is to provide a factory image (merged bin). What i have suggested too.

ItzArty, 2025-03-13:
> [@ItzArty](https://github.com/ItzArty) espwebtools uses esptool.js under the hood ;-) The solution is to provide a factory image (merged bin). What i have suggested too. btw micropython generates a combined firmware.bin https://github.com/micropython/micropython/blob/master/ports/esp32/README.md

I have investigated into esp-web-tools, but it turns out they don't actually alter the binaries in any way, yet, for some reason, it patches just fine? I'm a little puzzled here, I'm using the same binary, esp-web-tools, which after investigating (https://github.com/esphome/esp-web-tools/blob/6.1.1/src/flash.ts - also the version I'm using) doesn't actually alter it, works, but esptool-js doesn't. Any clue as to how this is even possible?

brianignacio5, 2025-03-18:
@drbdrb @Jason2866  I've added a PR to test a possible use of flash size "detect" 

https://github.com/espressif/esptool-js/pull/191

Could you please take a look ?

ItzArty, 2025-03-19:
Hello, I managed to fix this all by patching bytes 2 and 3 (index 2, 3) to 0x02 and 0x40, which solved the problem for me.
Thanks @Jason2866 for leading me to the correct path!