espressif/esptool-js #124: Browser module imports don't work

elite174, 2023-12-29:
[image]

I'm trying to use this lib in the browser without compilation and get this error.

elite174, 2023-12-29:
try the following code to reproduce
```
<!doctype html>
<html lang="en">
  <head>
    <meta charset="UTF-8" />
    <meta name="viewport" content="width=device-width, initial-scale=1.0" />
    <title>Document</title>
  </head>
  <body>
    <script type="module">
      import json from "https://unpkg.com/esptool-js@0.3.2/lib/targets/esp32.js?module";
    </script>
  </body>
</html>

```

elite174, 2023-12-29:
I guess the reason for this this line:
`import ESP32_STUB from "./stub_flasher/stub_flasher_32.json?module";`
The point is that you don't need to use `?module` for json files

elite174, 2023-12-29:
@brianignacio5 Could you fix this please?

brianignacio5, 2024-01-08:
How about using the resulting bundle file : https://unpkg.com/esptool-js@0.3.2/bundle.js ?

elite174, 2024-01-10:
Yes, that would work. But anyways, it's better be fixed

elite174, 2024-01-10:
Also you need to update README then, because it recommends using https://unpkg.com/esptool-js/lib/index.js?module

closed 2024-01-10 by elite174