espressif/esptool-js #125: package.json library decleration leads esbuild to incorrectly import index.js as a CommonJS file

hhvrc, 2024-01-24:
Due to the way its defined this makes the library unusable with esbuild projects like Vite, SvelkteKit as a npm module.

## Steps to reproduce:

### 1. Install project (with typescript support)
```
npm create svelte@latest my-app
cd my-app
npm i -D esptool-js @types/w3c-web-serial
```

### 2. Create example usage in any file
```
import { ESPLoader, Transport, type LoaderOptions } from 'esptool-js';

export async function ESPTest() {
    const port = await navigator.serial.requestPort();

    const transport = new Transport(port);

    const flashOptions = {
        transport,
        baudrate: 115200
    } as LoaderOptions;

    const loader = new ESPLoader(flashOptions);

    return { loader, transport };
}
```

### 3. Import this into +page.svelte and use it
```
<script lang="ts">
    import { ESPTest } from '$lib';
</script>

<p>test: {ESPTest()}</p>
```

### 4. Build project
```
npm run build
```

## Error output:
```
(node:30556) Warning: To load an ES module, set "type": "module" in the package.json or use the .mjs extension.
(Use `node --trace-warnings ...` to show where the warning was created)

node:internal/event_target:1100
  process.nextTick(() => { throw err; });
                           ^
F:\Development\Source\MyRepo\Frontend\node_modules\esptool-js\lib\index.js:1
export { ESPLoader } from "./esploader";
^^^^^^

SyntaxError: Unexpected token 'export'
    at internalCompileFunction (node:internal/vm:77:18)
    at wrapSafe (node:internal/modules/cjs/loader:1288:20)
    at Module._compile (node:internal/modules/cjs/loader:1340:27)
    at Module._extensions..js (node:internal/modules/cjs/loader:1435:10)
    at Module.load (node:internal/modules/cjs/loader:1207:32)
    at Module._load (node:internal/modules/cjs/loader:1023:12)
    at cjsLoader (node:internal/modules/esm/translators:356:17)
    at ModuleWrap.<anonymous> (node:internal/modules/esm/translators:305:7)
    at ModuleJob.run (node:internal/modules/esm/module_job:218:25)
    at async ModuleLoader.import (node:internal/modules/esm/loader:329:24)
Emitted 'error' event on Worker instance at:
    at [kOnErrorMessage] (node:internal/worker:326:10)
    at [kOnMessage] (node:internal/worker:337:37)
    at MessagePort.<anonymous> (node:internal/worker:232:57)
    at [nodejs.internal.kHybridDispatch] (node:internal/event_target:826:20)
    at exports.emitMessage (node:internal/per_context/messageport:23:28)
```

brianignacio5, 2024-01-31:
I've made a fix in #127. Could you test the artifact to see if it works ?

hhvrc, 2024-02-01:
Fix implemented in #127 does not fix my problem.

I am currently using [patch-package](https://github.com/ds300/patch-package) to fix esptool-js being broken locally.

This is the patch that makes esptool-js work for me:

```patch
diff --git a/node_modules/esptool-js/lib/esploader.js b/node_modules/esptool-js/lib/esploader.js
index 6e80537..09df456 100644
--- a/node_modules/esptool-js/lib/esploader.js
+++ b/node_modules/esptool-js/lib/esploader.js
@@ -1,8 +1,8 @@
-import { ESPError } from "./error";
+import { ESPError } from "./error.js";
 import { deflate, Inflate } from "pako";
-import { Transport } from "./webserial";
-import { customReset, usbJTAGSerialReset } from "./reset";
-import { Buffer } from "buffer/index";
+import { Transport } from "./webserial.js";
+import { customReset, usbJTAGSerialReset } from "./reset.js";
+import { Buffer } from "buffer/index.js";
 /**
  * Return the chip ROM based on the given magic number
  * @param {number} magic - magic hex number to select ROM.
diff --git a/node_modules/esptool-js/lib/index.js b/node_modules/esptool-js/lib/index.js
index e999db9..14d574f 100644
--- a/node_modules/esptool-js/lib/index.js
+++ b/node_modules/esptool-js/lib/index.js
@@ -1,4 +1,4 @@
-export { ESPLoader } from "./esploader";
-export { classicReset, customReset, hardReset, usbJTAGSerialReset, validateCustomResetStringSequence } from "./reset";
-export { ROM } from "./targets/rom";
-export { Transport } from "./webserial";
+export { ESPLoader } from "./esploader.js";
+export { classicReset, customReset, hardReset, usbJTAGSerialReset, validateCustomResetStringSequence } from "./reset.js";
+export { ROM } from "./targets/rom.js";
+export { Transport } from "./webserial.js";
```

hhvrc, 2024-02-27:
@brianignacio5 #127 does not fix this issue, can you re-open this?

Thanks

closed 2024-03-20 by brianignacio5

peter-pakanun, 2025-03-22:
I have the same issue using esptool-js@0.5.4 with svelte5, adding `"type": "module",` to esptool-js's `package.json` fixed it for me.

hhvrc, 2025-03-25:
@brianignacio5 yeah issue is still not fixed, I need to patch esptool-js with type: module for it to work

PR #127 (2024-01-31): update buffer ref in esploader
Was using a node reference in esploader to Buffer class instead of the buffer npm module added.

This update should add reference to node_modules/buffer instead.

Fix #125
PR #132 (2024-03-20): add js extensions use window atob
Fix #125

Fix #130

Add .js to import to fix issues with Svelte import.

Add function to try to use `window.atob` if available.

PTAL @balloob @hhvrc