espressif/esptool-js #135: Lolin S2 mini : Error: Invalid custom reset sequence

patrickgmail, 2024-04-01:
Hi,
I'm unable to flash mi device. Here is the result:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x80c2
Connecting..........Error: Invalid custom reset sequence
```

brianignacio5, 2024-05-28:
Are you using https://espressif.github.io/esptool-js/ or you are using esptool-js on your own project? 

If you are using your own project, which reset sequence are you using ? Are you implementing your own sequence string and how does it look like ?

EmileSpecialProducts, 2024-10-02:
I have the same problem on the [WEMOS LOLIN S2](https://www.wemos.cc/en/latest/s2/s2_mini.html). 
I also tested this on the [ https://espressif.github.io/esptool-js/](https://espressif.github.io/esptool-js/)
This is linked to this issue [https://github.com/esphome/esp-web-tools/issues/515](https://github.com/esphome/esp-web-tools/issues/515)

You can find my project [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler)
This also has a script for platformIO to create the Bin files.
Consol log of the  [ https://espressif.github.io/esptool-js/](https://espressif.github.io/esptool-js/)
```
xterm.min.js:7 Canvas2D: Multiple readback operations using getImageData are faster with the willReadFrequently attribute set to true. See: https://html.spec.whatwg.org/multipage/canvas.html#concept-canvas-will-read-frequently
e.1906.t._drawToCache @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
requestAnimationFrame
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
e.fire @ xterm.min.js:7
e.5435.t.parse @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
e._innerWrite @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
setTimeout
e.write @ xterm.min.js:7
e.8969.t.write @ xterm.min.js:7
e.4389.e.writeln @ xterm.min.js:7
writeLine @ index.ts:77
write @ esploader.js:202
info @ esploader.js:227
$b9be20b2c9078c9b$export$b0f7a6c745790308 @ esploader.js:188
$382e02c9bbd5d50b$var$connectButton.onclick @ index.ts:96Understand this warning
webserial.js:47 TRACE 1185.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
webserial.js:115 Write bytes
webserial.js:47 TRACE 1187.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
webserial.js:47 TRACE 1250.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
webserial.js:115 Write bytes
webserial.js:47 TRACE 1251.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
webserial.js:228 Read bytes
webserial.js:47 TRACE 1252.000 Read 102 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0c0010804 | ...... .........
    0007071220000000 00c0c00108040007 | .... ...........
    07122000000000c0 c001080400070712 | .. .............
    2000000000c0c001 0804000707122000 |  ............. .
    000000c0c0010804 0007071220000000 | ............ ...
    00c0c0010804                      | ......
webserial.js:234 Slip reader results
webserial.js:47 TRACE 1253.000 Read 12 bytes: 010804000707122000000000
webserial.js:47 TRACE 1253.000 command op:0x0a data len=4 wait_response=1 timeout=3.000 data=00100040        
[... 373 lines cut ...]
    a8a90991d0ff0c0a 8a9922490082c8c1 | .........."I....
    0c1980a983a08074 cc8aa2af40aa2220 | .......t....@." 
    98938cf910112025 f2ffc60100ad0281 | ...... %........
    efffe008001df036 4100a2a0dbdc1011 | .......6A.......
    20e5faff1df00000 36410082a0dbdcad |  .......6A......
    02879211a2a0dbdd 10112065f9ffa2a0 | .......... e....
    dc46040000000082 a0dbdd8792081011 | .F..............
    2025f8ffa2a0dd10 1120a5f7ff1df036 |  %....... .....6
    41003a32c60200a2 020022c201101120 | A.:2......".... 
    a5fbff3792f01df0 0000006c5200408c | ...7.......lR.@.
    7201408c5200400c 530040362121a2d1 | r.@.R.@.S.@6!!..
    1081faffe0080046 0b0000000c144044 | .......F......@D
    11404363cd04bd01 ad0281f5ffe00800 | .@Cc............
    a0a
index.ts:103 Error: Timeout
    at $4616b1a2f9a5b082$export$86495b081fef8e52.read (webserial.js:215:27)
    at async $b9be20b2c9078c9b$export$b0f7a6c745790308.readPacket (esploader.js:343:23)
    at async $b9be20b2c9078c9b$export$b0f7a6c745790308.checkCommand (esploader.js:573:22)
    at async $b9be20b2c9078c9b$export$b0f7a6c745790308.memBlock (esploader.js:607:9)
    at async $b9be20b2c9078c9b$export$b0f7a6c745790308.runStub (esploader.js:946:13)
    at async $b9be20b2c9078c9b$export$b0f7a6c745790308.main (esploader.js:1016:9)
    at async $382e02c9bbd5d50b$var$connectButton.onclick (index.ts:98:12)
$382e02c9bbd5d50b$var$connectButton.onclick @ index.ts:103Understand this error
index.ts:107 Settings done for :null
```

EmileSpecialProducts, 2024-10-05:
I have the same problem on the [WEMOS LOLIN S2](https://www.wemos.cc/en/latest/s2/s2_mini.html).
I also tested this on the https://espressif.github.io/esptool-js/
This is linked to this issue https://github.com/esphome/esp-web-tools/issues/515 and also #135 

You can find my project [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler)
The project can program the S2 But only you will need to use [this](https://www.emsign.nl/firmware/S2upload.html) 8.0.6 Webtool instaler as the 10.0.1 does not work on the S2 board.

This also has a script for platformIO to create the Bin files.

esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting....
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH32
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 80:65:99:eb:74:06
Uploading stub...
Error: Timeout

EmileSpecialProducts, 2024-10-05:
I have the same problem on the [WEMOS LOLIN S2](https://www.wemos.cc/en/latest/s2/s2_mini.html).
I also tested this on the https://espressif.github.io/esptool-js/
This is linked to this issue https://github.com/esphome/esp-web-tools/issues/515 and also #135 

You can find my project [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler)
The project **can** program the S2 But only you will need to use [this](https://www.emsign.nl/firmware/S2upload.html) old version 8.0.6 Webtool instaler as the new 10.0.1 does not work on the S2 board.

This also has a script for platformIO to create the Bin files.

esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting....
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH32
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 80:65:99:eb:74:06
Uploading stub...
Error: Timeout

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

EmileSpecialProducts, 2024-11-16:
I have tested the ESP32-S2(usb) and it still will not program the board.
I also have problems connecting the board that will also not always work. 

You can find my project [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler)
The project can program the S2 But only you will need to use [this](https://www.emsign.nl/firmware/S2upload.html) old version 8.0.6 Webtool instaler as the new 10.0.1 does not work on the S2 board.

esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic 7c6
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH32
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 48:27:e2:46:c4:5e
Uploading stub...
Debug: mem_begin 4444 1 6144 40028000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
**Error: No serial data received.**

TRACE 3172.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 3172.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 3622.000 No serial data received.
TRACE 3623.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 3623.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 3624.000 Read 1 bytes: c0              
TRACE 3625.000 Read 1 bytes: 01              
TRACE 3626.000 Read 1 bytes: 08              
TRACE 3626.000 Read 1 bytes: 04              
TRACE 3634.000 Read 1 bytes: 00              
TRACE 3634.000 Read 13 bytes: 0707122000000000c0c0010804
TRACE 3634.000 Received full packet: 010804000707122000000000
TRACE 3634.000 Read 4 bytes: c0010804        
TRACE 3635.000 Read 1 bytes: 00              
TRACE 3635.000 Read 1 bytes: 07              
TRACE 3635.000 Read 1 bytes: 07              
TRACE 3636.000 Read 1 bytes: 12              
TRACE 3636.000 Read 1 bytes: 20              
TRACE 3637.000 Read 1 bytes: 00              
TRACE 3638.000 Read 1 bytes: 00              
TRACE 3639.000 Read 1 bytes: 00              
TRACE 3640.000 Read 1 bytes: 00              
TRACE 3641.000 Read 1 bytes: c0              
TRACE 3641.000 Received full packet: 010804000707122000000000
TRACE 3641.000 Read 1 bytes: c0              
TRACE 3642.000 Read 1 bytes: 01              
TRACE 3642.000 Read 1 bytes: 08              
[... 776 lines cut ...]
    88dbdc89418826ad 0987b2010c3a99e1 | ....A.&......:..
    a9d1e9c110112065 1affa8d181e3fde8 | ...... e........
    c1a901a1e2fddd08 bd07c2c124f2c110 | ............$...
    89c181f5fde00800 b826cd0aa89198e1 | .........&......
    a0bbdbdcb926a022 dbdcb803aa77a841 | .....&.".....w.A
    88c1aabb0c0ab903 dbdca98380bbdbdc | ................
    a0d074cc9ae2dbdd 80ad0de0a98316ea | ..t.............
    01ad0889c199e1c9 d1101120e525ff88 | ........... .%..
    c198e1c8d1890346 010000000c1c9d0c | .......F........
    8cb238358c73dbdc 3f31dbdc33dbdc96 | ..85.s..?1..3...
    b3f5d68c0022a0c7 29558653fe0056ac | ....."..)U.S..V.
    94283516529422a0 c8c6faff282356a2 | .(5.R.".....(#V.
    93101120254cffa2 a271dbdcaa1181bc | ... %L...q......
    fde0080010112025 33ff81cefde00800 | ...... %3.......
    4646fe0028331632 91101120a549ffa2 | FF..(3.2... .I..
    a3e881b3fde00800 101120e530ffe002 | .......... .0...
    00063efe00101120 2530ff1df0000036 | ..>.... %0.....6
    41009d0282a0dbdc 280387990fcc320c | A.......(.....2.
    128607000c022903 7ce2860f00261207 | ......).|....&..
    2622188603000000 82a0dbdd80292387 | &"...........)#.
    992a0c2229037cf2 460800000022a0dc | .*.").|.F...."..
    27990a0c1229032d 08060400000082a0 | '....).-........
    dd7cf28799060c12 290322a0dbdd1df0 | .|......).".....
    0000c0                            | ...
TRACE 6980.000 No serial data received.

brianignacio5, 2024-11-21:
I think I may have found the culprit. Esp32s2 was missing the post connect function that update the ESP_RAM_BLOCK for S2 chips.

Changes are up in https://brianignacio5.github.io/esptool-js/ Could you please try again ? Apologies for the trouble.

EmileSpecialProducts, 2024-11-21:
No trouble.

Programming is not working the Write is at 100% in 0 ms as it looks.
you can find the Bin files at this [location](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler/tree/main/firmware/esp32s2_dio) 
What can i do to help resolve the issue?

Screen Log:
```
Esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v1.0)
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 80:65:99:eb:74:06
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Compressed 13792 bytes to 9687...
Took 0.0s to erase flash block
Writing at 0x1000... (100%)
Wrote 13792 bytes (9687 compressed) at 0x1000 in 0.07 seconds.
File  md5: 968dd94aec92e7194546429948b8ddf0
Flash md5: b3720e6eca62b07c9255c3978a088a1b
Error: MD5 of file does not match data in flash!
Erasing flash (this may take a while)...
Chip erase completed successfully in 14.931s

```
[image]

Trace:

> Trace
TRACE 3174.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 3175.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 3359.000 No serial data received.
TRACE 3359.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 3359.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 3360.000 Read 1 bytes: 00              
TRACE 3360.000 Read invalid data: 00              
TRACE 3360.000 Remaining data in serial buffer: 07              
TRACE 3360.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 3360.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 3360.000 Read 1 bytes: 07              
TRACE 3360.000 Read invalid data: 07              
TRACE 3360.000 Remaining data in serial buffer: 
    122000000000c0c0 0108040007071220 | . ............. 
[... 3479 lines cut ...]
    41f2cfef9c66f3ae 5c48dff0c5807ada | A....f..\H....z.
    7cc7b4b06c7a5a3e 327275f58cfe41b5 | |...lzZ>2ru...A.
    e928cbdd41a41a4e 7686714b4889a187 | .(..A..Nv.qKH...
    64083131402f22b4 3e172d3504d77bca | d.11@/".>.-5..{.
    74358d5e3ac1a591 0a6e81f87babd848 | t5.^:....n..{..H
    559d8d74a9d355e3 7534912d4215343b | U..t..U.u4.-B.4;
    9f651359969d095b 50be45a1b587d9b2 | .e.Y...[P.E.....
    b34672345b721e51 ae7c41468ce72e4e | .Fr4[r.Q.|AF...N
    b9e74e3b77aa5de1 7b6fc96ad5705ad1 | ..N;w.].{o.j.pZ.
    c914378be4dd71d3 827a9a89667eb47a | ..7...q..z..f~.z
    c6e9304e23ae5faa 4503b6aab8284255 | ..0N#._.E....(BU
    7d577b09a34ed5e3 e15c84680c335dff | }W{..N...\.h.3].
    ccf6ddfdaeb0c29d fd4f54d685ddb3ae | .........OT.....
    f64bcc2152677b97 85d1f21bd84e2527 | .K.!Rg{......N%'
    931678a7829b9a22 51936efb0a768bda | ..x...."Q.n..v..
    8f82feeea4393fe1 dbddcf5fdf439a78 | .....9?...._.C.x
    398ca39a67f3ea6a 9c4de7aa26cf4e38 | 9...g..j.M..&.N8
    f79c3335ce84357f f62e430d3e35795a | ..35..5...C.>5yZ
    de27eba41bfedbdc 5d121b7f2935e6fe | .'......]...)5..
    45a85a7f50b7b423 9ee9639955dea7b0 | E.Z.P..#..c.U...
    e4d765df49399961 66e08ba9cfbf98fa | ..e.I9.af.......
    84c96881c772f8ec d4dbdc2753c22753 | ..h..r.....'S.'S
    3dbcde5e889c8d33 6181e6d75124cc31 | =..^...3a...Q$.1
    a317c849b5774657 87a9eb826cf478ff | ...I.wFW....l.x.
    1bbffffa4ca

Theaninova, 2024-11-23:
Same issue, though with the branch mine seems to get a bit further?

```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH2 (revision v0.0)
Features: Wi-Fi,Embedded Flash 2MB,No Embedded Flash,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 70:04:1d:48:1a:5e
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Compressed 22928 bytes to 14695...
Took 0.0s to erase flash block
Writing at 0x1000... (100%)
Wrote 22928 bytes (14695 compressed) at 0x1000 in 0.063 seconds.
Hash of data verified.
Compressed 152016 bytes to 99796...
Took 0.0s to erase flash block
Writing at 0x190000... (14%)
Writing at 0x199f1f... (28%)
Writing at 0x19f739... (42%)
Writing at 0x1a4e34... (57%)
Writing at 0x1aa1bd... (71%)
Writing at 0x1af307... (85%)
Writing at 0x1b47d7... (100%)
Wrote 152016 bytes (99796 compressed) at 0x190000 in 0.404 seconds.
Hash of data verified.
Compressed 3072 bytes to 137...
Took 0.0s to erase flash block
Writing at 0x8000... (100%)
Wrote 3072 bytes (137 compressed) at 0x8000 in 0.006 seconds.
Hash of data verified.
Compressed 8192 bytes to 47...
Took 0.0s to erase flash block
Writing at 0xe000... (100%)
Wrote 8192 bytes (47 compressed) at 0xe000 in 0.01 seconds.
File  md5: e6327541e2dc394ca2c3b3280ac0f39f
Flash md5: e7a15f9ca2a901424ca56bbe9b6987b4
Error: MD5 of file does not match data in flash!
```
[trace.txt](https://github.com/user-attachments/files/17882534/trace.txt)

brianignacio5, 2024-11-25:
Hi @EmileSpecialProducts  I've found that the IS_STUB was not being set so the flash values were not being picked correctly. Now I fixed it.

Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

EmileSpecialProducts, 2024-11-25:
I have mange to Program the ESP32-S2 Lolin S2 Mini.
It is a lot faster then the Old version.

[image]

esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic 7c6
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v1.0)
Features: Wi-Fi,Embedded Flash 4MB,Embedded PSRAM 2MB,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 80:65:99:eb:74:06
Debug: In _post_connect using USB OTG ?true
Uploading stub...
Debug: mem_begin 4444 3 2048 40028000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Debug: check_command write to target RAM
Debug: check_command write to target RAM
Debug: mem_begin 160 1 2048 3ffe2bfc
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Running stub...
Debug: check_command leave RAM download mode
Stub running...
Changing baudrate to 921600
Changed
Debug: EspLoader program
Debug: Data Length 13792
Debug: Image Length 13792
Debug: _update_image_flash_params keep undefined undefined
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Debug: Image MD5 968dd94aec92e7194546429948b8ddf0
Compressed 13792 bytes to 9687...
Debug: check_command enter compressed flash mode
Debug: Write loop 4096 0 1
Writing at 0x1000... (100%)
Debug: flash_defl_block 78 da
Debug: check_command write compressed data to flash after seq 0
Wrote 13792 bytes (9687 compressed) at 0x1000 in 0.255 seconds.
Debug: check_command calculate md5sum
Hash of data verified.
Debug: Data Length 3072
Debug: Image Length 3072
Debug: _update_image_flash_params keep undefined undefined
Debug: Image MD5 a039c66cd3488176037b616b7595fe72
Compressed 3072 bytes to 146...
Debug: check_command enter compressed flash mode
Debug: Write loop 32768 0 1
Writing at 0x8000... (100%)
Debug: flash_defl_block 78 da
Debug: check_command write compressed data to flash after seq 0
Wrote 3072 bytes (146 compressed) at 0x8000 in 0.043 seconds.
Debug: check_command calculate md5sum
Hash of data verified.
Debug: Data Length 8192
Debug: Image Length 8192
Debug: _update_image_flash_params keep undefined undefined
Debug: Image MD5 e6327541e2dc394ca2c3b3280ac0f39f
Compressed 8192 bytes to 47...
Debug: check_command enter compressed flash mode
Debug: Write loop 57344 0 1
Writing at 0xe000... (100%)
Debug: flash_defl_block 78 da
Debug: check_command write compressed data to flash after seq 0
Wrote 8192 bytes (47 compressed) at 0xe000 in 0.086 seconds.
Debug: check_command calculate md5sum
Hash of data verified.
Debug: Data Length 283136
Debug: Image Length 283136
Debug: _update_image_flash_params keep undefined undefined
Debug: Image MD5 0a0aafdc793ecfd934091aee29fb8e9a
Compressed 283136 bytes to 163738...
Debug: check_command enter compressed flash mode
Debug: Write loop 65536 0 10
Writing at 0x10000... (10%)
Debug: flash_defl_block 78 da
Debug: check_command write compressed data to flash after seq 0
Debug: Write loop 65536 1 10
Writing at 0x1b701... (20%)
Debug: flash_defl_block fc 7f
Debug: check_command write compressed data to flash after seq 1
Debug: Write loop 65536 2 10
Writing at 0x2442c... (30%)
Debug: flash_defl_block f7 90
Debug: check_command write compressed data to flash after seq 2
Debug: Write loop 65536 3 10
Writing at 0x297f7... (40%)
Debug: flash_defl_block fc c2
Debug: check_command write compressed data to flash after seq 3
Debug: Write loop 65536 4 10
Writing at 0x2e9fb... (50%)
Debug: flash_defl_block fd 3e
Debug: check_command write compressed data to flash after seq 4
Debug: Write loop 65536 5 10
Writing at 0x33d65... (60%)
Debug: flash_defl_block 4e 84
Debug: check_command write compressed data to flash after seq 5
Debug: Write loop 65536 6 10
Writing at 0x39148... (70%)
Debug: flash_defl_block bc f4
Debug: check_command write compressed data to flash after seq 6
Debug: Write loop 65536 7 10
Writing at 0x41758... (80%)
Debug: flash_defl_block c d4
Debug: check_command write compressed data to flash after seq 7
Debug: Write loop 65536 8 10
Writing at 0x4a27d... (90%)
Debug: flash_defl_block 85 d3
Debug: check_command write compressed data to flash after seq 8
Debug: Write loop 65536 9 10
Writing at 0x4f771... (100%)
Debug: flash_defl_block 19 36
Debug: check_command write compressed data to flash after seq 9
Wrote 283136 bytes (163738 compressed) at 0x10000 in 2.259 seconds.
Debug: check_command calculate md5sum
Hash of data verified.
Leaving...
Debug: flash begin 0 0 16384 0 0
Debug: check_command enter Flash download mode
Debug: check_command leave compressed flash mode

brianignacio5, 2024-11-26:
Hi @Theaninova Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

Theaninova, 2024-11-27:
The device just reboots after the _connect_attempt, I also tried it with `no_reset` and same result.
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 3
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 4
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: _connect_attempt default_reset
Error: Failed to execute 'setSignals' on 'SerialPort': Failed to set control signals.
```

I should note that the device already has firmware on it, I'm not sure if a device without firmware would reboot to bootloader fast enough to where the serial connection did not get interrupted? I also used the 1200 baud connection to get it into bootloader before if that is relevant.

brianignacio5, 2024-11-27:
Which device are you using @Theaninova ? It seems that setDTR or setRTS are not being sent so reset is not being done and sync is not being done.

This is using the default reset strategy from esptool.py. Does it work with esptool.py ? Could you share the tracing output ?

Theaninova, 2024-11-27:
I was using a [CharaChorder Lite S2](https://www.charachorder.com/en-de/products/charachorder-lite)

```
❯ esptool.py --port /dev/ttyACM6 flash_id
esptool.py v4.7.0
Serial port /dev/ttyACM6
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FH4 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:db:52:18
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...
```

[esptoolpy-trace.txt](https://github.com/user-attachments/files/17933682/esptoolpy-trace.txt)

closed 2024-11-28 by brianignacio5

The1Percent, 2024-12-21:
> Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

This works!

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]