espressif/esptool-js #140: Monitor for ESP32-C6

WeeJeWel, 2024-04-30:
If I'm not mistaken, the ESP32-C6 (and the ESP32-C3) can enable a Serial Monitor by sending a command to the bootloader. Afterwards, serial output (e.g. `Serial.println("Hello!")`) could then be received inside a browser.

Can esptool-js also support this?

brianignacio5, 2024-05-28:
You can try https://espressif.github.io/esptool-js/ and Click the Start button under console section for monitoring.

WeeJeWel, 2024-05-28:
That didn’t work, hence this issue.

brianignacio5, 2024-05-29:
[image]

This is using the ESP32-C6-WROOM-1.

brianignacio5, 2024-05-30:
I'm using the ESP32-C6-DevKit-C1 so I'm using the type C USB UART bridge as shown here:

https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html#hardware-reference

WeeJeWel, 2024-05-30:
Thank you. This issue applies to the ESP32-C6 built-in USB controller, not a devkit's UART bridge.

brianignacio5, 2024-05-30:
Also works with the USB port.

Could you provide the tracing information from the Developer Tools or the connect button ?

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

EmileSpecialProducts, 2024-11-27:
It work by me, ESP32-C6 dev board can use the reset on CH343 and USB.
Set USB CDC On Boot 'Enabled'

I have used this to test it 
```
#include <Arduino.h> //  https://github.com/espressif/arduino-esp32/tree/master/cores/esp32
void setup() {
// set ARDUINO_USB_CDC_ON_BOOT
Serial.begin(115200);
Serial.setDebugOutput(true);
Serial0.begin(115200);
Serial0.setDebugOutput(true);

}
int counter=0;
void loop() {
Serial.println("Serial "+String(counter));
Serial0.println("Serial0 "+String(counter));
counter++;
sleep(1);
}
```

brianignacio5, 2024-12-16:
Closing this issue due to lack of response from original user, please comment here if you need more assistance.

closed 2024-12-16 by brianignacio5