espressif/esptool-js #141: Timeout connecting, but works on adafruit version of esptool-js

tyeth, 2024-05-09:
Board = PyPortal Titano (SAMD51 with onboard ESP32)
Uploaded the [SerialNINAPassthru sketch](https://github.com/adafruit/WiFiNINA/blob/master/examples/Tools/SerialNINAPassthrough/SerialNINAPassthrough.ino) to allow esp32 to be connected to serial port in download mode.
Connecting to serial shows esp boot message and download mode.
[image]

Connecting using esptool-js site just keeps attempting to connect and fails https://espressif.github.io/esptool-js/
[image]

Connecting using adafruit site works. https://adafruit.github.io/Adafruit_WebSerial_ESPTool/
[image]

tyeth, 2024-05-09:
Getting a slip read error programming, going to try another pc.

tyeth, 2024-05-09:
Kept getting slip errors on adafruit esptool (espressif esptool wouldnt even connect) writing a 1.4mb firmware until I erased another time. (Had previously erased in theory, but not sure I let it complete). Then wrote fine

Still would expect espressif esptool-js to report the connected device like the adafruit one does.

brianignacio5, 2024-05-28:
Could you Copy the trace and share here ?

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

brianignacio5, 2024-11-20:
It seems the esp32 s2 has some issues yet. I have here a esp32 s2 devkitC 1 which seems to connect properly on my setup but I need more testing.

Could you try with esptool.py but using the `--trace` argument so I can compare the output ?

tyeth, 2024-11-20:
Just to confirm, which version of esptool.py?

tyeth, 2024-11-20:
This was 5.3.1 idf using esptool 4.8.1
<details><summary>Details</summary>
<p>

<pre>
C:\dev\esp-idf [(v5.3.1) +0 ~19 -0 !]> ./export.ps1
Setting IDF_PATH: C:\dev\esp-idf
Checking Python compatibility
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp-elf found in PATH: esp-13.2.0_20230928. To use it, run 'C:\Python312\python.exe C:\dev\esp-idf\tools\idf_tools.py export --prefer-system'

Name                           Value
----                           -----
OPENOCD_SCRIPTS                C:\dev\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\sh… 
IDF_CCACHE_ENABLE              1
ESP_ROM_ELF_DIR                C:\dev\Espressif\tools\esp-rom-elfs\20240305\
IDF_PYTHON_ENV_PATH            C:\dev\Espressif\python_env\idf5.3_py3.12_env
ESP_IDF_VERSION                5.3
IDF_DEACTIVATE_FILE_PATH       C:\Users\tyeth\AppData\Local\Temp\tmp8n_1h77qidf_4584

Added to PATH
-------------
C:\dev\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
C:\dev\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
C:\dev\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin
C:\dev\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin
C:\dev\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin
C:\dev\Espressif\tools\cmake\3.24.0\bin
C:\dev\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin
C:\dev\Espressif\tools\ninja\1.11.1\
C:\dev\Espressif\tools\idf-exe\1.0.3\
C:\dev\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
C:\dev\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
C:\dev\Espressif\python_env\idf5.3_py3.12_env\Scripts
C:\dev\esp-idf\tools
Checking if Python packages are up to date...
Constraint file: C:\dev\Espressif\espidf.constraints.v5.3.txt
Requirement files:
 - C:\dev\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\dev\Espressif\python_env\idf5.3_py3.12_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of espup, msys2, rustup, vc_redist, xtensa-esp32s2-elf use command 'python.exe C:\dev\esp-idf\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\dev\esp-idf\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

C:\dev\esp-idf [(v5.3.1) +0 ~19 -0 !]> esptool.py.exe
esptool.py.exe: The term 'esptool.py.exe' is not recognized as a name of a cmdlet, function, script file, or executable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
C:\dev\esp-idf [(v5.3.1) +0 ~19 -0 !]> python -m esptool --trace erase_flash                                 
esptool.py v4.8.1
Found 1 serial ports
Serial port COM29
Connecting...
COM29 failed to connect: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)

A fatal error occurred: Could not connect to an Espressif device on any of the 1 available serial ports.     
C:\dev\esp-idf [(v5.3.1) +0 ~19 -0 !]> python -m esptool --trace erase_flash
esptool.py v4.8.1
Found 1 serial ports
Serial port COM7
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
[... 1092 lines cut ...]
TRACE +0.003 command op=0xd0 data len=0 wait_response=1 timeout=120.000 data=
TRACE +0.000 Write 10 bytes: c000d0000000000000c0
TRACE +0.001 Read 1 bytes: c0
TRACE +16.291 Read 1 bytes: 01
TRACE +0.000 Read 10 bytes: d00200000000000000c0
TRACE +0.001 Received full packet: 01d00200000000000000
Chip erase completed successfully in 16.3s
TRACE +0.004 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=3840403f
TRACE +0.000 Write 14 bytes: c0000a0400000000003840403fc0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 10 bytes: 0a0200000000000000c0
TRACE +0.000 Received full packet: 010a0200000000000000
TRACE +0.003 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=2881403f
TRACE +0.000 Write 14 bytes: c0000a0400000000002881403fc0
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 10 bytes: 0a0200010000000000c0
TRACE +0.000 Received full packet: 010a0200010000000000
Hard resetting via RTS pin...
C:\dev\esp-idf [(v5.3.1) +0 ~19 -0 !]> 
</pre>

</p>
</details>

brianignacio5, 2024-11-20:
Hi @tyeth  thank you for the very detailed feedback. I've continued to do more changes and released them in my fork. There was some code in the run stub that was not aligned to esptool.py and new changes should make it more similar. Please take a look again. I hope to resolve these issues you are facing as soon as possible.

Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

tyeth, 2024-11-20:
Debug log version:
```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 3
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic 7c6
Detecting chip type... ESP32-S2
Chip is ESP32-S2 (revision v0.0)
Features: Wi-Fi,No Embedded Flash,No Embedded Flash,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:d9:b1:46
Uploading stub...
Debug: mem_begin 4444 1 6144 40028000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Error: No serial data received.
```

Browser console log:
[esptool-js-log-20241120_1429.txt](https://github.com/user-attachments/files/17832096/esptool-js-log-20241120_1429.txt)

tyeth, 2024-11-20:
@brianignacio5 I should probably mention that I'm now in a state where there is no normal serial port when running, so I'm using the bootloader stage1 reset mode to see the device in the serial port list (so the device reset before stub upload is technically unneccessary as it's already in download mode). Not sure if that makes a difference for your code/test

brianignacio5, 2024-11-21:
I think I may have found the culprit. Esp32s2 was missing the post connect function that update the ESP_RAM_BLOCK for S2 chips.

Changes are up in https://brianignacio5.github.io/esptool-js/ Could you please try again ? Apologies for the trouble.

tyeth, 2024-11-21:
@brianignacio5 Seems to connect nicely now, and erase was successful (repeatedly in theory), but writing a small file at 0x0 had a weird pause at 60% and then finished, but failed checksum verification of the write.
Writing a different file at 0x10000 the checksum also fails after writing.

<details><summary>Debug log Details</summary>
<p>
<pre>
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x2
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 3
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic 7c6
Detecting chip type... ESP32-S2
Chip is ESP32-S2 (revision v0.0)
Features: Wi-Fi,No Embedded Flash,No Embedded Flash,ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: 60:55:f9:d9:b1:46
Debug: In _post_connect using USB OTG ?true
Uploading stub...
Debug: mem_begin 4444 3 2048 40028000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Debug: check_command write to target RAM
Debug: check_command write to target RAM
Debug: mem_begin 160 1 2048 3ffe2bfc
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Running stub...
Debug: check_command leave RAM download mode
Stub running...
Changing baudrate to 921600
Changed
Erasing flash (this may take a while)...
Debug: check_command erase flash
Chip erase completed successfully in 16.273s
Debug: EspLoader program
Debug: Data Length 3113904
Debug: Image Length 3113904
Debug: _update_image_flash_params keep undefined undefined
Debug: Image MD5 b6a27342b3b68de9954807f586561251
Compressed 3113904 bytes to 323130...
Debug: check_command enter compressed flash mode
Took 0.0s to erase flash block
Debug: Write loop 0 0 20
Writing at 0x0... (5%)
Debug: flash_defl_block 78 da
Debug: check_command write compressed data to flash after seq 0
Debug: Write loop 0 1 20
Writing at 0x11000... (10%)
Debug: flash_defl_block e3 ed
Debug: check_command write compressed data to flash after seq 1
Debug: Write loop 0 2 20
Writing at 0x1d24d... (15%)
Debug: flash_defl_block c1 18
[... 521 lines cut ...]
Writing at 0x2e9958... (90%)
Debug: flash_defl_block 9d 17
Debug: check_command write compressed data to flash after seq 17
Debug: Write loop 0 18 20
Writing at 0x2eeb5a... (95%)
Debug: flash_defl_block 1c d5
Debug: check_command write compressed data to flash after seq 18
Debug: Write loop 0 19 20
Writing at 0x2f3e51... (100%)
Debug: flash_defl_block a6 36
Debug: check_command write compressed data to flash after seq 19
Wrote 3113904 bytes (323130 compressed) at 0x0 in 3.525 seconds.
Debug: check_command calculate md5sum
File  md5: b6a27342b3b68de9954807f586561251
Flash md5: 0fa1d49fe2452360d9cc730dba2715e2
Error: MD5 of file does not match data in flash!
</pre>
</p>
</details> 

Browser Console Log Details:
[esptool-js-log-20241121_1344.txt](https://github.com/user-attachments/files/17846776/esptool-js-log-20241121_1344.txt)

Firmware being written:
[Metro_ESP32S2_FactoryReset (1).zip](https://github.com/user-attachments/files/17846779/Metro_ESP32S2_FactoryReset.1.zip)

tyeth, 2024-11-21:
@brianignacio5 I haven't got any of the Adafruit ESP32-C3 QtPy's and I'm told we wont be producing more, but I grabbed a random SuperMini C3 that I got from aliexpress, and it erases fine at 921k baud, but the write fails verification. Same at 115k baud

brianignacio5, 2024-11-25:
Hi @tyeth   I've found that the IS_STUB was not being set so the flash values were not being picked correctly. Now I fixed it.

Could you please retry my fork in https://brianignacio5.github.io/esptool-js/ with Show Debug log enabled ?

tyeth, 2024-11-25:
@brianignacio5 looks good on the Metro S2 that I have handy, erase and verify successful, there was a big pause in writing the factory reset image at about 70% - no idea why, I'll retest the others later.

There was a considerable amount of weirdness with not opening the serial port (it was always in bootloader mode and visible), which then breaks the disconnect button, requiring the tab to be refreshed. I found that resetting the device (boot0 + reset) and refreshing the tab resolved the issue.  Maybe the connected state could be not enabled if the device is still null (as it shows in the UI label).

The serial errors on connect varied between couldn't fire setDTR/RTS, and the "Failed to execute 'open' on 'SerialPort': Failed to open serial port.". I'll try to keep an eye on it, and restart my poor machine before labelling it buggy 😉

Console just shows:
```
esptool-js/:22 Canvas2D: Multiple readback operations using getImageData are faster with the willReadFrequently attribute set to true. See: https://html.spec.whatwg.org/multipage/canvas.html#concept-canvas-will-read-frequently
apply @ esptool-js/:22
getImageData @ esptool-js/:242
t._drawToCache @ xterm.min.js:7
t.draw @ xterm.min.js:7
e._drawChars @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
t._forEachCell @ xterm.min.js:7
t._drawForeground @ xterm.min.js:7
t.onGridChanged @ xterm.min.js:7
t.renderRows @ xterm.min.js:7
t._renderRows @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
e._innerRefresh @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
requestAnimationFrame
e.refresh @ xterm.min.js:7
t.refreshRows @ xterm.min.js:7
t.refresh @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
e.fire @ xterm.min.js:7
t.parse @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
e._innerWrite @ xterm.min.js:7
(anonymous) @ xterm.min.js:7
setTimeout
e.write @ xterm.min.js:7
t.write @ xterm.min.js:7
e.writeln @ xterm.min.js:7
writeLine @ index.ts:80
write @ esploader.js:185
info @ esploader.js:210
$b9be20b2c9078c9b$export$b0f7a6c745790308 @ esploader.js:171
$382e02c9bbd5d50b$var$connectButton.onclick @ index.ts:100Understand this warningAI
14:55:47.264 index.ts:107 NetworkError: Failed to execute 'open' on 'SerialPort': Failed to open serial port.
$382e02c9bbd5d50b$var$connectButton.onclick @ index.ts:107Understand this errorAI
14:55:47.265 index.ts:111 Settings done for :null
14:55:53.998 webserial.js:379 Uncaught (in promise) InvalidStateError: Failed to execute 'close' on 'SerialPort': The port is already closed.
    at $4616b1a2f9a5b082$export$86495b081fef8e52.disconnect (webserial.js:379:27)
    at async $382e02c9bbd5d50b$var$disconnectButton.onclick (index.ts:219:18)
disconnect @ webserial.js:379
await in disconnect
$382e02c9bbd5d50b$var$disconnectButton.onclick @ index.ts:219Understand this errorAI
14:55:57.202 Navigated to https://brianignacio5.github.io/esptool-js/
```

tyeth, 2024-11-27:
Thanks Brian!
I think it's just that connections which fail still then show a connected state (device listed as null in page and the 3 buttons show including disconnect), and then clicking disconnect errors silently or endlessly waits due to the await port something at the top of the disconnect routine. I'll give you a better report in future (on phone now)

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]
closed 2024-11-28 by brianignacio5