espressif/esptool-js #142: When the flash is erased and the power is accidentally cut off, the connection will be stuck in "connecting..."

HenTaiCJN, 2024-05-15:
Under normal circumstances, the connection prompt is 
```shell
connecting....._____....._____
```
 This may take a long time, but at least the connection will be successful.

When the flash is erased and the power is accidentally cut off, the connection will be stuck in "connecting...". 
```javascript
esploader.js
 async connect(mode = "default_reset", attempts = 7, detecting = false) {
        let i;
        let resp;
        this.info("Connecting...", false);
        await this.transport.connect(this.romBaudrate, this.serialOptions);
        for (i = 0; i < attempts; i++) {
            resp = await this._connectAttempt(mode, false);
            if (resp === "success") {
                break;
            }
            resp = await this._connectAttempt(mode, true);
            if (resp === "success") {
                break;
            }
        }
``` 
I speculate that the program is stuck at 'await this.transport.connect(this.romBaudrate, this.serialOptions);'，but I don't know how to solve it.
I tried to flash the firmware using esptool.exe, and esptool.exe worked normally. So the device itself should be fine.

brianignacio5, 2024-05-28:
Which board or chip are you using ?

Could you copy the trace ? Which Baud rate are you using ?

HenTaiCJN, 2024-05-30:
> Which board or chip are you using ?
> 
> Could you copy the trace ? Which Baud rate are you using ?

Chip is ESP32-D0WD-V3 (revision v3.0)
Baudrate is 115200
trace keeps printing a lot of the same stuff, filling up my console

```consolog
webserial.js:234 Slip reader results
webserial.js:47 TRACE 12653.000 Read 0 bytes:                 
webserial.js:228 Read bytes
webserial.js:47 TRACE 7726.000 Read 6626 bytes:
657473204a756c20 3239203230313920 | ets Jul 29 2019
31323a32313a3436 0d0a0d0a7273743a | 12:21:46....rst:
3078312028504f57 45524f4e5f524553 | 0x1 (POWERON_RES
4554292c626f6f74 3a30783132202853 | ET),boot:0x12 (S
50495f464153545f 464c4153485f424f | PI_FAST_FLASH_BO
4f54290d0a657473 204a756c20323920 | OT)..ets Jul 29
323031392031323a 32313a34360d0a0d | 2019 12:21:46...
0a7273743a307831 2028504f5745524f | .rst:0x1 (POWERO
4e5f524553455429 2c626f6f743a3078 | N_RESET),boot:0x
313220285350495f 464153545f464c41 | 12 (SPI_FAST_FLA
53485f424f4f5429 0d0a696e76616c69 | SH_BOOT)..invali
6420686561646572 3a203078666666666 | d header: 0xffff
666666660d0a696e 76616c6964206865 | ffff..invalid he
616465723a203078 66666666666666666 | ader: 0xffffffff
0d0a696e76616c69 6420686561646572 | ..invalid header
3a20307866666666 666666660d0a696e | : 0xffffffff..in
76616c6964206865 616465723a203078 | valid header: 0x
666666666666666 0d0a696e76616c69 | ffffffff..invali
6420686561646572 3a203078666666666 | d header: 0xffff
666666660d0a696e 76616c6964206865 | ffff..invalid he
616465723a203078 66666666666666666 | ader: 0xffffffff
0d0a696e76616c69 6420686561646572 | ..invalid header
3a20307866666666 666666660d0a696e | : 0xffffffff..in
76616c6964206865 616465723a203078 | valid header: 0x
66666666666666666 0d0a657473204a75 | ffffffff..ets Ju
6c20323920323031 392031323a32313a | l 29 2019 12:21:
34360d0a0d0a7273 743a307831302028 | 46....rst:0x10 (
5254435744545f52 54435f5245534554 | RTCWDT_RTC_RESET
292c626f6f743a30 7831322028535049 | ),boot:0x12 (SPI
5f464153545f464c 4153485f424f4f54 | _FAST_FLASH_BOOT
290d0a696e76616c 6964206865616465 | )..invalid heade
723a203078666666 66666666660d0a69 | r: 0xffffffff..i
6e76616c69642068 65616465723a2030 | nvalid header: 0
7866666666666666 660d0a696e76616c | xffffffff..inval
6964206865616465 723a203078666666 | id header: 0xfff
66666666660d0a69 6e76616c69642068 | fffff..invalid h
65616465723a2030 78666666666666666 | edge: 0xfffffff
660d0a696e76616c 6964206865616465 | f..invalid heade
723a203078666666 66666666660d0a69 | r: 0xffffffff..i
6e76616c69642068 65616465723a2030 | nvalid header: 0
7866666666666666 660d0a696e76616c | xffffffff..inval
6964206865616465 723a203078666666 | id header: 0xfff
66666666660d0a69 6e76616c69642068 | fffff..invalid h
65616465723a2030 78666666666666666 | edge: 0xfffffff
660d0a657473204a 756c203239203230 | f..ets Jul 29 20
31392031323a3231 3a34360d0a0d0a72 | 19 12:21:46....r
73743a3078313020 285254435744545f | st:0x10 (RTCWDT_
5254435f52455345 54292c626f6f743a | RTC_RESET),boot:
3078313220285350 495f464153545f46 | 0x12 (SPI_FAST_F
4c4153485f424f4f 54290d0a696e7661 | LASH_BOOT)..inva
```

brianignacio5, 2024-11-28:
Could you please try again with https://espressif.github.io/esptool-js/ ? IN #160 many changes were implemented to improve connection stability.

HenTaiCJN, 2024-11-28:
> Could you please try again with https://espressif.github.io/esptool-js/ ? IN #160 many changes were implemented to improve connection stability.

Oh, it looks much more stable. Even if the power is cut off during the flash erasing, it can be reconnected. I will test it further later. Thank you.

brianignacio5, 2025-03-20:
I'm closing this issue. Comment here if you need more assistance.

closed 2025-03-20 by brianignacio5