espressif/esptool-js #143: How can I have predefined files from a Github Release?

AllanOricil, 2024-05-19:
For example, this [Release](https://github.com/AllanOricil/esp32-mfa-totp-generator/releases/tag/v0.6.0) has a build for the CYD. How can I configure esptool-js to with the files and proper flash indexes?

Thank you in advance.

brianignacio5, 2024-05-28:
Go to https://espressif.github.io/esptool-js/ and click Connect under Program section. 

Add binaries (.bin) and flash offset address in the list. The list of binaries to flash and the flash offset address are usually defined in the partitions table.

Click Program. See the flashing output. If successful you can disconnect

AllanOricil, 2024-05-28:
I'm looking for a way to build a custom site where people can just click on a button called "flash", or "update", to flash their boards easily. @brianignacio5

brianignacio5, 2024-05-29:
One thing you can do is to prepare a TOML file with your binaries and call the ESP-Launchpad with the URL (like GitHub for example) of your TOML and it can be used for flashing monitor etc.

Here is the link to the ESP-Launchpad tutorial in Elektor magazine: https://www.elektormagazine.com/magazine/elektor-328/62435

Here is the ESP Launchpad repository: https://github.com/espressif/esp-launchpad

Note: This project uses esptool-js as dependency so it is doing the same thing but simplify your binaries configuration.

AllanOricil, 2024-05-29:
@brianignacio5 one last question. Do you know a way I can pass parameters while flashing the board with my build? For example, during builds I can set WIFI_SSID and WIFI_PASSWORD as build variables. However I can't ship firmwares with these values hard-coded. I need the user to set them up when flashing.

AllanOricil, 2024-05-29:
Never mind. I think I will change my code to read those env specific variables at runtime from a config file.

DrSkunk, 2024-08-13:
@AllanOricil Maybe ESP Web Tools suit your usecase better https://esphome.github.io/esp-web-tools/

AllanOricil, 2024-08-13:
> @AllanOricil Maybe ESP Web Tools suit your usecase better https://esphome.github.io/esp-web-tools/

@DrSkunk thank you

Im actually using it already :D

This is my project
https://github.com/AllanOricil/esp32-mfa-authenticator

closed 2024-08-13 by AllanOricil