espressif/esptool-js #150: Error: Timeout if esp32-s3 is already programmed

abarangan, 2024-08-07:
On a mac running Chrome, while connecting to an esp32-s3 via esptool-js seems to work fine, attempting to program an esp32-s3 that is already running an application (built with esp-idf 5.2.2) is always returning Error: Timeout. If I locally erase flash with idf.py and then run esptool-js, flashing is successful. My desired use case for esptool-js doesn't allow me to first erase flash. Here is the trace.

TRACE 335.000 Read 151 bytes: 
    4553502d524f4d3a 657370333273332d | ESP-ROM:esp32s3-
    3230323130333237 0d0a4275696c643a | 20210327..Build:
    4d61722032372032 3032310d0a727374 | Mar 27 2021..rst
    3a30783135202855 53425f554152545f | :0x15 (USB_UART_
    434849505f524553 4554292c626f6f74 | CHIP_RESET),boot
    3a3078302028444f 574e4c4f41442855 | :0x0 (DOWNLOAD(U
    53422f5541525430 29290d0a53617665 | SB/UART0))..Save
    642050433a307834 303033346634300d | d PC:0x40034f40.
    0a77616974696e67 20666f7220646f77 | .waiting for dow
    6e6c6f61640d0a                    | nload..
TRACE 335.000 Read 0 bytes:                 
TRACE 1389.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 1390.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 1442.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 1442.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 1442.000 Read 263 bytes: 
    4553502d524f4d3a 657370333273332d | ESP-ROM:esp32s3-
    3230323130333237 0d0a4275696c643a | 20210327..Build:
    4d61722032372032 3032310d0a727374 | Mar 27 2021..rst
    3a30783135202855 53425f554152545f | :0x15 (USB_UART_
    434849505f524553 4554292c626f6f74 | CHIP_RESET),boot
    3a3078302028444f 574e4c4f41442855 | :0x0 (DOWNLOAD(U
    53422f5541525430 29290d0a53617665 | SB/UART0))..Save
    642050433a307834 303033346634300d | d PC:0x40034f40.
    0a77616974696e67 20666f7220646f77 | .waiting for dow
    6e6c6f61640d0ac0 0108040007071220 | nload.......... 
    00000000c0c00108 0400070712200000 | ............. ..
    0000c0c001080400 0707122000000000 | ........... ....
    c0c0010804000707 122000000000c0c0 | ......... ......
    0108040007071220 00000000c0c00108 | ....... ........
    0400070712200000 0000c0c001080400 | ..... ..........
    0707122000000000 c0c0010804000707 | ... ............
    122000000000c0                    | . .....
TRACE 1442.000 Read 12 bytes: 010804000707122000000000
TRACE 1442.000 command op:0x0a data len=4 wait_response=1 timeout=3.000 data=00100040        
TRACE 1443.000 Write 14 bytes: c0000a04000000000000100040c0
TRACE 1443.000 Read 28 bytes: 
    c001080400070712 2000000000c0c001 | ........ .......
    0804000707122000 000000c0         | ...... .....
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1443.000 Read 14 bytes: c0010804000707122000000000c0
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1443.000 Read 14 bytes: c0010804000707122000000000c0
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1443.000 Read 14 bytes: c0010804000707122000000000c0
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1443.000 Read 14 bytes: c0010804000707122000000000c0
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1443.000 Read 14 bytes: c0010804000707122000000000c0
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1443.000 Read 14 bytes: c0010804000707122000000000c0
TRACE 1443.000 Read 12 bytes: 010804000707122000000000
TRACE 1444.000 Read 14 bytes: c0010a04000900000000000000c0
TRACE 1444.000 Read 12 bytes: 010a04000900000000000000
TRACE 1444.000 command op:0x0a data len=4 wait_response=1 timeout=3.000 data=44700060        
[... 706 lines cut ...]
    40b58e3740968d37 40bc913740369337 | @..7@..7@..7@6.7
    40728c3740569337 40728c3740728c37 | @r.7@V.7@r.7@r.7
    40728c3740728c37 40728c3740728c37 | @r.7@r.7@r.7@r.7
    40728c3740728c37 4056913740728c37 | @r.7@r.7@V.7@r.7
    4051923740369337 4004089f00000000 | @Q.7@6.7@.......
    0000001801040805 0000000000000008 | ................
    0104080600000000 0000000001040821 | ...............!
    0000000000200000 110408dc00000000 | ..... ..........
    002000001104080c 0000000000200000 | . ........... ..
    0104081200000000 002000001120280c | ......... ... (.
    0010010000c0                      | ......
TRACE 1477.000 Read 14 bytes: c0010704000400000000000000c0
TRACE 1477.000 Read 12 bytes: 010704000400000000000000
TRACE 1477.000 command op:0x06 data len=8 wait_response=1 timeout=0.050 data=00000000408a3740
TRACE 1477.000 Write 18 bytes: 
    c000060800000000 0000000000408a37 | .............@.7
    40c0                              | @.
TRACE 1478.000 Read 14 bytes: c0010604000400000000000000c0
TRACE 1478.000 Read 12 bytes: 010604000400000000000000
TRACE 1482.000 Read 6 bytes: c04f484149c0    
TRACE 1482.000 Read 4 bytes: 4f484149        
TRACE 16146.000 command op:0x10 data len=16 wait_response=1 timeout=3.000 data=00300000010000000040000000000000
TRACE 16146.000 Write 26 bytes: 
    c000101000000000 0000300000010000 | ..........0.....
    0000400000000000 00c0             | ..@.......

jakobofner, 2024-09-17:
Hi,
I have the same Problem. I think it is related to the the process setting the esp in programming mode. 
Therefore I'd suggest adapting the connection sequence which is defined in esploader.ts on line 611:

const strSequence = esp32r0Delay ? "D0|R1|W100|W2000|D1|R0|W50|D0" : "D0|R1|W100|D1|R0|W50|D0";

I was not able to find a suitable working for my purpose too.

jakobofner, 2024-09-17:
I found a solution for the problem which is working for my hardware:

I commented line 5 in esploader.ts:
`import { ESPError } from "./error.js";
import { Data, deflate, Inflate } from "pako";
import { Transport, SerialOptions } from "./webserial.js";
import { ROM } from "./targets/rom.js";
//import { customReset, usbJTAGSerialReset } from "./reset.js";
import atob from "atob-lite";`

Changed the _connectAttempt function to the following:

`async _connectAttempt(mode = "default_reset", esp32r0Delay = false) {
    this.debug("_connect_attempt " + mode + " " + esp32r0Delay);
    if (mode !== "no_reset") {
      await this.transport.setDTR(false); // EN LOW (Reset)
      await this.transport.setRTS(true); // EN LOW (Reset)
      
      await new Promise((resolve)=>setTimeout(resolve, 100));
      
      await this.transport.setDTR(true); // EN LOW (Reset)
      await this.transport.setRTS(false); // EN LOW (Reset)
      
      await new Promise((resolve)=>setTimeout(resolve, 50));
      
      await this.transport.setDTR(false); // EN LOW (Reset)
    }
    let i = 0;
    let keepReading = true;
    while (keepReading) {
      try {
        const res = await this.transport.read(1000);
        i += res.length;
      } catch (e) {
        this.debug((e as Error).message);
        if (e instanceof Error) {
          keepReading = false;
          break;
        }
      }
      await this._sleep(50);
    }
    this.transport.slipReaderEnabled = true;
    this.syncStubDetected = true;
    i = 7;
    while (i--) {
      try {
        const resp = await this.sync();
        this.debug(resp[0].toString());
        return "success";
      } catch (error) {
        if (error instanceof Error) {
          if (esp32r0Delay) {
            this.info("_", false);
          } else {
            this.info(".", false);
          }
        }
      }
      await this._sleep(50);
    }
    return "error";
  }`
  
  in the esptool-js folder run the following commands to test it:
  
  `npm install
  npm run build
  cd examples/typescript/
  npm install
  npm run dev`

brianignacio5, 2024-11-13:
Could you try https://brianignacio5.github.io/esptool-js/ which implement #160 ?

jakobofner, 2024-11-13:
Hi, with your Implementation it works now fine when I change connect baudrate to 115200! Thanks!

abarangan, 2024-11-18:
Also works here Brian. Flashed my esp32-s3 at 921600 baud. Thanks!

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]
closed 2024-11-18 by abarangan