espressif/esptool-js #159: esp32s3固件烧录失败 Error：Timeout

TMOW, 2024-10-31:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

串口通讯正常，固件烧录曾经是正常的

### Actual behavior (suspected bug)

两周前测试可以正常烧录固件，今天测试时发现无法烧录，但是错误提示不全，只有Error：Timeout
node的插件版本："esptool-js": "^0.4.5"
报错截图和使用的固件在压缩包内

### Error logs or terminal output

```shell
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE,Embedded PSRAM 2MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:f2:0a:d8
Uploading stub...
Running stub...
Stub running...
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 1967200 bytes to 1287022...
Error: Timeout
```

### Steps to reproduce the behavior

只需要烧录就可以复现

### Project release version

node插件版本0.4.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win10

### Shell

CMD

brianignacio5, 2024-11-01:
Could you try enabling tracing and share the trace output here ?
tracing is enabled using the transport class second argument as shown here: https://github.com/espressif/esptool-js/blob/main/examples/typescript/src/index.ts#L89

Example to get trace in here: https://github.com/espressif/esptool-js/blob/main/examples/typescript/src/index.ts#L125

TMOW, 2024-11-04:
我做了进一步测试，如果板子当前的固件是从0x0地址开始烧录的，那么在下一次烧录固件时，会卡在切换波特率这里，我做了bug排查，是在esploader.ts文件的1224行代码：const resp = await this.command(this.ESP_CHANGE_BAUDRATE, pkt)；[https://github.com/espressif/esptool-js/blob/f699520/src/esploader.ts#L1224](url)
我是用一块esp32s3在从0x0地址烧录成功后，第二次烧录任意地址时会出现的报错信息如下：
[image]

brianignacio5, 2024-11-05:
I've been trying to reproduce your issue without success:

I'm using esp32s3 DevKitC-1 v1.0 with current github esptool-js website without issues:
[image]

TMOW, 2024-11-05:
我使用的硬件是esp32-s3-box
[image]

这个是我测试用的固件，需要从0x0地址开始烧录，在第二次就会出问题

[test.zip](https://github.com/user-attachments/files/17628430/test.zip)

TMOW, 2024-11-05:
这是我的录屏，我在第一次从0x0地址烧录时，情况正常，但是当我第二次开始从0x0地址烧录时就会出现问题

https://github.com/user-attachments/assets/4f384f7b-6acb-4e71-97ca-dd31e7032bf2

brianignacio5, 2024-11-05:
Could you press Copy Trace button on the 2nd flashing and share the output here ? Output should be copied in the clipboard

TMOW, 2024-11-05:
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting...Debug: _connect_attempt default_reset false
Debug: Timeout
Debug: Sync
Debug: Sync err Error: Timeout
.Debug: Sync
Debug: 538052359

Debug: Chip Magic 9
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE,Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:e1:a5:9c
Debug: In _post_connect 4
Uploading stub...
Debug: mem_begin 5076 1 6144 40378000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Debug: mem_begin 252 1 6144 3fcb2bf4
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Running stub...
Debug: check_command leave RAM download mode
Stub running...
Changing baudrate to 921600
Error: Timeout
Debug: EspLoader program
Debug: Data Length 1967200
Debug: Image Length 1967200
Debug: _update_image_flash_params keep undefined undefined
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Debug: Image MD5 f0d111676c30cc5904ee5b30e3d76863
Compressed 1967200 bytes to 1287022...
Debug: check_command enter compressed flash mode
Error: Timeout

brianignacio5, 2024-11-06:
Could you try again but changing the baud rate to 460800 before connect ?

I managed to connect and flash using a lower baud rate than 921600

TMOW, 2024-11-06:
我尝试了所有的波特率，全部都失败了
这是460800
[image]
这是230400
[image]
这是115200
[image]

brianignacio5, 2024-11-06:
Using Program baud rate 460800  and the usb port I got this:
[image]

```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE,Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:e1:55:94
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 460800
Changed
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 1967200 bytes to 1287022...
Writing at 0x0... (1%)
Writing at 0x125c2... (2%)
Writing at 0x19ec6... (3%)
Writing at 0x217fa... (5%)
Writing at 0x29d2d... (6%)
Writing at 0x31359... (7%)
Writing at 0x37ff5... (8%)
Writing at 0x3f2c1... (10%)
Writing at 0x495d4... (11%)
Writing at 0x4f745... (12%)
Writing at 0x5abbb... (13%)
Writing at 0x611fb... (15%)
Writing at 0x66884... (16%)
Writing at 0x6b278... (17%)
Writing at 0x743cb... (18%)
Writing at 0x7c727... (20%)
Writing at 0x84307... (21%)
Writing at 0x8f2fd... (22%)
Writing at 0x949b8... (24%)
Writing at 0x9a289... (25%)
Writing at 0x9f8a9... (26%)
Writing at 0xa48d1... (27%)
Writing at 0xa99ae... (29%)
Writing at 0xae97e... (30%)
Writing at 0xb3b7f... (31%)
Writing at 0xb8a63... (32%)
Writing at 0xbd7ac... (34%)
Writing at 0xc2815... (35%)
Writing at 0xc9251... (36%)
Writing at 0xce4e4... (37%)
Writing at 0xd42a6... (39%)
Writing at 0xd9d64... (40%)
Writing at 0xdf357... (41%)
Writing at 0xe4a84... (43%)
Writing at 0xea17e... (44%)
Writing at 0xef6f7... (45%)
Writing at 0xf4f8e... (46%)
Writing at 0xfa72d... (48%)
Writing at 0xffd5b... (49%)
Writing at 0x1052c4... (50%)
Writing at 0x10a623... (51%)
Writing at 0x10fc3c... (53%)
Writing at 0x114f5a... (54%)
Writing at 0x11a1c4... (55%)
Writing at 0x11f316... (56%)
Writing at 0x124bce... (58%)
Writing at 0x12abdd... (59%)
Writing at 0x130e4b... (60%)
Writing at 0x136084... (62%)
Writing at 0x13ae4b... (63%)
Writing at 0x13ff8d... (64%)
Writing at 0x14526d... (65%)
Writing at 0x14a7d1... (67%)
Writing at 0x14ffd6... (68%)
Writing at 0x154f37... (69%)
Writing at 0x15a136... (70%)
Writing at 0x15f9ec... (72%)
Writing at 0x1648fd... (73%)
Writing at 0x169e7d... (74%)
Writing at 0x16f44c... (75%)
Writing at 0x1745bc... (77%)
Writing at 0x17a0bf... (78%)
Writing at 0x17ef9d... (79%)
Writing at 0x184074... (81%)
Writing at 0x1890fa... (82%)
Writing at 0x18e3ba... (83%)
Writing at 0x193710... (84%)
Writing at 0x198fc4... (86%)
Writing at 0x19e598... (87%)
Writing at 0x1a3d64... (88%)
Writing at 0x1a9cdb... (89%)
Writing at 0x1af8a8... (91%)
Writing at 0x1b8c9f... (92%)
Writing at 0x1c06df... (93%)
Writing at 0x1c6986... (94%)
Writing at 0x1cc0a0... (96%)
Writing at 0x1d1b63... (97%)
Writing at 0x1d722f... (98%)
Writing at 0x1dd08a... (100%)
Wrote 1967200 bytes (1287022 compressed) at 0x0 in 22.81 seconds.
Hash of data verified.
Leaving...
```

TMOW, 2024-11-06:
你有没有手动进入下载模式？我在测试时发现，按下硬件上的按键主动进入下载模式时是可以正常烧录的，如果不手动进入下载模式就会失败，统一出现在Changing baudrate to xxxxxx时，会出现timeout，定位在代码的这一行：https://github.com/espressif/esptool-js/blob/f69952001ffe5f2ef469ee9ab00da0c530a1ef55/src/esploader.ts#L1224
[image]

brianignacio5, 2024-11-06:
I don't press any button in the esp32s3. I just use the esptool-js website and that's it.

Did you modified the boot loader mode somehow ? https://docs.espressif.com/projects/esptool/en/latest/esp32s3/advanced-topics/boot-mode-selection.html

brianignacio5, 2024-11-06:
yes, I can flash multiple times the test.bin you shared.

TMOW, 2024-11-06:
我觉得如果boot loader mode是有问题的，你在测试时使用了test固件，那么你应该会跟我一样出现多次烧录时会失败的问题

TMOW, 2024-11-06:
我使用esptool.exe 在test.bin固件上进行了多次烧录测试，不会出现切换波特率失败的情况
[image]

C:\Users\Administrator>D:\github\mPython-vue\labplus-link\tools\esptool.exe -p COM29 -b 4000000 -c esp32s3 write_flash 0x0000 E:\work\ESP32-S3\固件\test.bin
esptool.py v4.8.1
Serial port COM29
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:e1:a5:9c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 4000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x001e0fff...
Compressed 1967200 bytes to 1287022...
Wrote 1967200 bytes (1287022 compressed) at 0x00000000 in 22.0 seconds (effective 716.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

C:\Users\Administrator>D:\github\mPython-vue\labplus-link\tools\esptool.exe -p COM29 -b 4000000 -c esp32s3 write_flash 0x0000 E:\work\ESP32-S3\固件\test.bin
esptool.py v4.8.1
Serial port COM29
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:e1:a5:9c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 4000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x001e0fff...
Compressed 1967200 bytes to 1287022...
Wrote 1967200 bytes (1287022 compressed) at 0x00000000 in 22.0 seconds (effective 715.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

TMOW, 2024-11-07:
我们找了6台电脑进一步测试，3台win10系统和3台win11系统，只有其中一台win11系统的是可以正常使用的，其他5台均无法多次烧录固件

brianignacio5, 2024-11-13:
Could you try https://brianignacio5.github.io/esptool-js/ which implement #160 ?

brianignacio5, 2024-11-13:
I'll take a look at the reset flow. Thank you for the quick test

TMOW, 2024-11-13:
我在测试的时候发现有概率出现bug
[image]
log如下：
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting...Debug: using USB JTAG Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x53): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x53): Possible serial noise or corruption.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x20): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x20): Possible serial noise or corruption.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x70): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x70): Possible serial noise or corruption.
Debug: Sync connect attempt 3
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0xa): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0xa): Possible serial noise or corruption.
Debug: Sync connect attempt 4
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x28): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x28): Possible serial noise or corruption.
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x6d): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x6d): Possible serial noise or corruption.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x5b): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x5b): Possible serial noise or corruption.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x30): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x30): Possible serial noise or corruption.
Debug: Sync connect attempt 3
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x0): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x0): Possible serial noise or corruption.
Debug: Sync connect attempt 4
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Error: Invalid head of packet (0x45): Possible serial noise or corruption.

brianignacio5, 2024-11-13:
Based on the output you can see that is reading an invalid packet because the esp32s3 has not reset properly and not yet in boot loader mode.

Could you describe me the steps to reproduce this behavior ?

TMOW, 2024-11-13:
我测试了一下，是我硬件的固件有问题，在正常的固件下不会出现这个问题

TMOW, 2024-11-14:
大概多久后这次更新会推送到npm上？

TMOW, 2024-11-14:
现在都正常了

TMOW, 2024-11-27:
我把这个（https://github.com/espressif/esptool-js/pull/160）内的0.4.7版本的文件通过npm install 安装到我的项目中了
[image]
但是仍然会出现烧录固件时timeout的情况
[image]
这是我的烧录固件的代码
[image]

brianignacio5, 2024-11-27:
Version 0.4.7 doesn't include #160 so it doesn't fix your issue.

I will release v0.5.0 soon I'm just waiting for user feedback for bug fixes.

TMOW, 2024-11-27:
从Pull request 下载的

brianignacio5, 2024-11-27:
Could you set the `debugLogging` flag to true on ESPLoader creation ?

```ts
const flashOptions = {
      transport,
      baudrate: 921600,
      debugLogging: true,
    } as LoaderOptions;
    esploader = new ESPLoader(flashOptions);
```

Need more output information. It could also be related to the bundle, you can try consuming the index.js directly to see if the problem lies within the bundle.

TMOW, 2024-11-28:
这是我的报错日志：
Port closed.
bundle.js:2 esptool.js
bundle.js:2 Serial port WebSerial VendorID 0x303a ProductID 0x1001
bundle.js:2 Connecting...
bundle.js:2 Debug: _connect_attempt default_reset false
bundle.js:2 Debug: Timeout
bundle.js:2 Debug: Sync
bundle.js:2 Debug: Sync err Error: Timeout
bundle.js:2 .
bundle.js:2 Debug: Sync
bundle.js:2 Debug: 538052359
bundle.js:2 

bundle.js:2 Debug: Chip Magic 9
bundle.js:2 Detecting chip type... 
bundle.js:2 ESP32-S3
bundle.js:2 Chip is ESP32-S3
bundle.js:2 Features: Wi-Fi,BLE,Embedded PSRAM 2MB (AP_3v3)
bundle.js:2 Crystal is 40MHz
bundle.js:2 MAC: 7c:df:a1:f2:0a:90
bundle.js:2 Debug: In _post_connect 4
bundle.js:2 Uploading stub...
bundle.js:2 Debug: mem_begin 5076 1 6144 40378000
bundle.js:2 Debug: check_command enter RAM download mode
bundle.js:2 Debug: check_command write to target RAM
bundle.js:2 Debug: mem_begin 252 1 6144 3fcb2bf4
bundle.js:2 Debug: check_command enter RAM download mode
bundle.js:2 Debug: check_command write to target RAM
bundle.js:2 Running stub...
bundle.js:2 Debug: check_command leave RAM download mode
bundle.js:2 Stub running...
bundle.js:2 Changing baudrate to 921600
port.js:441 Error: Timeout
    at Ce.read (bundle.js:2:49023)
    at async Re.readPacket (bundle.js:2:55045)
    at async Re.changeBaud (bundle.js:2:67473)
    at async Re.main (bundle.js:2:68291)

brianignacio5, 2024-11-28:
Could you try using index.js instead of the bundle.js ?

TMOW, 2024-11-28:
是我的本地缓存问题，现在可以了

PR #160 (2024-11-28): update serial slip read
Fix #159 
Fix #150 
Fix #151
Fix #38
Fix #135 
Fix #141
Fix #86 
Fix #87

## Description

Re-write the transport read and slip parsing to match esptool.py. This method should remove the timeout issue seen in most reported issue.

Update stubs to match current esptool.py stubs. Change baud issue in #159 was probably due to stub entry not matching.

## Testing

You can test on my fork in https://brianignacio5.github.io/esptool-js/

Or test by `npm run dev` the **examples/typescript** after `npm run build` in root directory. This will generate a localhost version of https: [... cut]
closed 2024-11-28 by brianignacio5