espressif/esptool-js #161: ESP32-C3 issues

diplfranzhoepfinger, 2024-11-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

.

### Actual behavior (suspected bug)

.

### Error logs or terminal output

```shell
.
```

### Steps to reproduce the behavior

.

### Project release version

.

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

.

### Shell

ZSH

### Additional context

.

brianignacio5, 2024-11-15:
Could you try my fork in https://brianignacio5.github.io/esptool-js/ which implement #160 changes ?

EmileSpecialProducts, 2024-11-16:
I have tested a ESP32-C3 DIO board looks lit it is programming ok, but the software will not run.
you can find de Turtle-ESP-mouse-jiggler firmeware files at this [location](https://github.com/EmileSpecialProducts/EmileSpecialProducts.github.io/tree/main/Turtle-ESP-mouse-jiggler/firmware) 
and the Turtle-ESP-mouse-jiggler software itself at this [location](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler).

EmileSpecialProducts, 2024-11-27:
Hi, It looks like it is programming the ESP32-C3,
But the code will not start and the C3 reboots continasly, 
I added the logfile 
[2024-11-27-ESPC3 log not programming.txt](https://github.com/user-attachments/files/17939201/2024-11-27-ESPC3.log.not.programming.txt)

[image]
you can find de Turtle-ESP-mouse-jiggler firmeware files at this [location](https://github.com/EmileSpecialProducts/EmileSpecialProducts.github.io/tree/main/Turtle-ESP-mouse-jiggler/firmware)
and the Turtle-ESP-mouse-jiggler software itself at this [location](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler).

tyeth, 2024-12-05:
@EmileSpecialProducts 
I've just tried your software, on a MakerGo C3 Supermini, using my version of esptool-js which includes Brians changes (now in main branch), along with the partition reading ability. See here: https://tyeth.github.io/read-partitions-esptool-js/
I've erased first, then applied the partition bin at 0x8000, and then the bootloader at 0x0, boot app partition at 0xe000 and firmware.bin at 0x10000, and the device is booted and reporting power on reason. 

It is **not boot looping**. My chip is reported by esptool as a C3 revision 4. It's also connected via BLE to my mac, but no idea what I'm doing with it so my test ends there.
Serial console reported:
```
Reboot from: Turtle ESP BLE ESP32-C3 CpuFreqMHz: 80 Total heap: 278 Free heap: 179 Total PSRAM: 0 Free PSRAM: 0 Temperature: 46.90 °C  FlashChipSize: 4194304 FlashChipSpeed: 40000000 FlashChipMode: FM_DIO Mac: 64:E8:33:88:52:94 reset_reason: Vbat power onBLE Connected
E (30082) BT_GATT: GATT_INSUF_AUTHENTICATION
```

EmileSpecialProducts, 2024-12-06:
[tyeth](https://github.com/tyeth) I have test your version on the C3 Supper mini and it will program en the [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler) code and will run.
So that issue is resolved.
I like the **partition reading ability**

brianignacio5, 2024-12-06:
@EmileSpecialProducts @tyeth  the esp32 C3 issue you were having was related to the app itself or esptool-js ? If the flashing is correct it shouldn't be but I wanted to get your feedback on that.

EmileSpecialProducts, 2024-12-06:
[brianignacio5](https://github.com/brianignacio5) I have just tested[ your](https://brianignacio5.github.io/esptool-js/) version And the same Bin files look like they are programming but the Software will not start but reports invalid handel.
My conclusion is that it is somthing in the esptool-js.

brianignacio5, 2024-12-06:
I will try to reproduce here with esp32c3-devkitM-1 which is the one I have here. I could try to get a esp32c3 DIO board but need some days. Do you have a link of the board you are using  ? So I make sure to get the right one.

EmileSpecialProducts, 2024-12-06:
[brianignacio5](https://github.com/brianignacio5) I now tested it on the C3 Supper minim that is just a plan ESP32-C3-4N Chip and see the same issue. so you no not need a DIO board, when you have a DIO software it will run on a QIO Chip/board only will not be as fast as a QIO compiled software.
when i used the Dump from [tyeth](https://github.com/tyeth) then i see that the images are diffrent. ( yours and Tyeth are the same) 
When i Program one big file ( generated from Tyeth read flash) then i can program it And it works.

tyeth, 2024-12-06:
@brianignacio5 I had no problems programming the C3. 

I did have a bit of an issue with reading data (was timing out part way through) when reading the whole flash chip using my fork, on the S2 and another chip, which I had only seen occasionally before, but maybe that was a crap usb cable (it goes through a hub) or unstable computer (Mac running beta's). Either way unrelated.

On a similar related but unrelated note, it would be much appreciated if there could be a release made, so that the cdn UnPkg get's it's scripts updated. (We - adafruit - are holding off on updating our ESPtool due to the S2 stub issue until there is a new release, but we've added the no-reset option to support our M4 + ESP32 coprocessor boards like the PyPortal so want to get it out)

EmileSpecialProducts, 2024-12-07:
I think the issue is that when i Program the C3 in 4 parts it will not program correctly.
When i have just on Big bin file it works.
[brianignacio5](https://github.com/brianignacio5) when you have a Relase candidate, i will run a test on all the Chips i have and report on them. as my the [Turtle-ESP-mouse-jiggler](https://github.com/EmileSpecialProducts/Turtle-ESP-mouse-jiggler) will support ESP32 /C3/C6/S2/S3/H2

brianignacio5, 2024-12-09:
I've tested with esp32c3 devkitm-1 and got this in the console output:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6a0
load:0x403ce710,len:0x25f4
entry 0x403cc710
```

Could you tell me what is the expected output ?

I was looking into the code. It seems there is some padding to the image based on image length:

```ts
const reminder = options.fileArray[i].data.length % 4;
 if (reminder > 0) image += "\xff\xff\xff\xff".substring(4 - reminder);
```

An invalid header might occur if the previous image has appended this values at beginning of bootloader. A single bin file would not show this issue since it would append it at the end. 

Does it work correctly using esptool.py ?

tyeth, 2024-12-09:
Amazing, thanks Brian!
Also a heads up on the read partitions code, it's a copilot assisted port of the gen_esp32part.py, still probably missing some functionality from the original file, and full of to-do/commented blocks 😇 more of a inspiration to do better than a complete solution, but it does decode partition tables 😊

brianignacio5, 2024-12-10:
I just pushed a new release to npm (v0.5.2) due to some publish issues I had before. PTAL @tyeth

About @EmileSpecialProducts apologies I wanted to fix this issue before but need to push work forward. Feel free to try my fork with the PR #166 to see if these fix the issues or we can keep looking the problem.

EmileSpecialProducts, 2024-12-11:
Test performed 2024-12-11 
Software [https://brianignacio5.github.io/esptool-js/](https://brianignacio5.github.io/esptool-js/) , 
And the 
 [firmeware](https://github.com/EmileSpecialProducts/EmileSpecialProducts.github.io/tree/main/Turtle-ESP-mouse-jiggler/firmware)
All tests has done on the following devices:

C3 Supper mini  Interface USB, Separate Files 			Status Ok 
C3 LUATOS Interface USB, Separate Files	 			Status Ok 
C3 LUATOS Interface CH343, Separate Files 			        Status Ok 
C6 Dev module Interface USB, Separate Files			        Status Ok
C6 Dev module Interface CH343, Separate Files			Status Ok
S3 Dev module Interface USB, Separate Files			        Status Ok
S3 Dev module Interface CH343, Separate Files			Status Ok
S2 WEMOS LOLIN S2 PICO Interface USB, Separate Files	Status Ok
H2 Dev module Interface USB, Separate Files			Status Ok
H2 Dev module Interface CH343, Separate Files			Status Ok
ESP32 Dev mod ESP32-D0WDQ6 (revision 1) 			Status Ok

Great work done!!!

tyeth, 2024-12-11:
@brianignacio5 just retested the bundle. Looks good (for the s2), so I'd imagine the chip json objects all got fixed up. Nice work!

brianignacio5, 2024-12-16:
Closing this issue due to lack of response from original user. Please try latest version and comment here if you have any more issues.

closed 2024-12-16 by brianignacio5