espressif/esptool-js #162: Javascript example for Progamimng

EmileSpecialProducts, 2024-11-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Can you add an example in JavaScript that will program a device.
Now I use the [ESP-Web-Tools](https://esphome.github.io/esp-web-tools/) but that will always be the current ESPtool.

### Use cases

Use case is recovery of a device is the OTA update has failed

### Alternatives

use the [ESP-Web-Tools](https://esphome.github.io/esp-web-tools/)

brianignacio5, 2024-11-28:
There is an example that is used to publish for GitHub pages in `examples/typescript`. 

The live also has link to documentation.

brianignacio5, 2025-03-13:
Closing this issue since example is already there. Comment here if you need more assistance.

closed 2025-03-13 by brianignacio5