espressif/esptool-js #167: 0.5.2 stub issue with cdn/npm bundle.js - stub object properties are now behind additional nested key of `default`

tyeth, 2024-12-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

stub uploads

### Actual behavior (suspected bug)

atob issue with stub data

### Error logs or terminal output

```shell
locally npm build fails with atob-lite export issue.

lib/index.js → bundle.js...
(!) Unresolved dependencies
https://rollupjs.org/troubleshooting/#warning-treating-module-as-external-dependency
fs (imported by "lib/gen_esp32part.py.js")
crypto (imported by "lib/gen_esp32part.py.js")
[!] RollupError: "default" is not exported by "node_modules/atob-lite/atob-browser.js", imported by "lib/stubFlasher.js".
https://rollupjs.org/troubleshooting/#error-name-is-not-exported-by-module
lib/stubFlasher.js (1:7)
1: import atob from "atob-lite";
          ^
2: /**
3:  * Import flash stub json for the given chip name.
```

using the individual js files works fine, using the bundle doesnt.
browser console error is:
```
script.js:177 InvalidCharacterError: Failed to execute 'atob' on 'Window': The string to be decoded is not correctly encoded.
    at Me (bundle.js:2:54124)
    at Fe (bundle.js:2:55198)
    at Qe (bundle.js:2:55140)
    at async fe.runStub (bundle.js:2:71082)
    at async fe.main (bundle.js:2:72633)
    at async clickConnect (script.js:168:16)
clickConnect @ script.js:177
await in clickConnect
(anonymous) @ script.js:31Understand this errorAI
17:44:39.240 script.js:108 Failed to execute 'atob' on 'Window': The string to be decoded is not correctly encoded.
```
```

### Steps to reproduce the behavior

checkout https://tyeth.github.io/read-partitions-esptool-js/

### Project release version

0.5.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

mac latest beta

### Shell

ZSH

tyeth, 2024-12-10:
[image]

tyeth, 2024-12-10:
Hey @brianignacio5 this one slipped through the net, I'd been using the split files version locally and on my site, similar to how esptool-js github pages is, but now that we've used the bundle.js in this PR the new version has broken something in the bundling, and the stubs are all behind an extra level of nesting (behind a key of `default:`).

brianignacio5, 2024-12-11:
Please take a look at https://github.com/espressif/esptool-js/pull/168

PR #168 (2024-12-12): rollup use namedExports
Fix #167 

Rollup seems to be packing the json in a weird. Hoping this PR fix the issue.
closed 2024-12-12 by brianignacio5