espressif/esptool-js #169: Unable to run dev on local windows host

Snehaaa-C, 2024-12-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I followed the setup instructions to run the ESP tool locally on my Windows but I'm encountering an issue where the localhost is not starting as expected.
[image]
[image]

### Actual behavior (suspected bug)

I followed the setup instructions to run the ESP tool locally on my Windows but I'm encountering an issue where the localhost is not starting as expected.  Could anyone kindly assist me in debugging this issue? I would greatly appreciate guidance on how to troubleshoot the problem. 

### Error logs or terminal output

```shell
PS C:\Users\SnehaC\Documents\esptool\esptool-js\examples\typescript> npm run dev                                                     

> typescript@1.0.0 dev
> npm run clean && npm run genDocs && parcel src/index.html

> typescript@1.0.0 clean
> rimraf dist .parcel-cache

> typescript@1.0.0 genDocs
> cd ../.. && npm run genDocs && mkdir -p examples/typescript/dist && cp -r docs examples/typescript/dist && cd examples/typescript  

> esptool-js@0.5.2 genDocs
> rimraf docs && typedoc

[warning] ResetStrategy, defined in ./src/reset.ts, is referenced by ClassicReset but not included in the documentation.
[warning] FlashReadCallback, defined in ./src/esploader.ts, is referenced by ESPLoader.readFlash.readFlash.onPacketReceived but not included in the documentation.
[info] Documentation generated at ./docs
The syntax of the command is incorrect.
PS C:\Users\SnehaC\Documents\esptool\esptool-js\examples\typescript>
```

### Steps to reproduce the behavior

1. set the environment.
2. install package
3. run command

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

11

### Shell

Bash

naveenneog, 2024-12-14:
Same issue happening with me

brianignacio5, 2024-12-16:
Instead of running `npm run dev` could you try running `npm run clean` `npm run genDocs` `parcel src/index.html` separately to find the failing command. Seems to me is probably the last one.

I can't reproduce it myself locally.

Snehaaa-C, 2024-12-16:
Thank you for your response!!

have tried to run each command separately the error was in genDocs it was not compatible with windows,
We changed it to windows compatible it is running now.

brianignacio5, 2024-12-16:
Oh sure. What you ended up with ? I can try to update it so it works on both windows/posix systems

Snehaaa-C, 2024-12-17:
We have changed the genDocs path command in scripts, below is the one which is updated package.json file.

{
  "name": "typescript",
  "version": "1.0.0",
  "description": "This an example of using esptool-js with parcel and typescript",
  "source": "src/index.html",
  "scripts": {
    "genDocs": "cd ..\\.. && npm run genDocs && mkdir examples\\typescript\\dist && xcopy docs examples\\typescript\\dist /E /I && cd examples\\typescript",
    "dev": "npm run clean && npm run genDocs && parcel src\\index.html",
    "build": "npm run clean && npm run genDocs && parcel build src\\index.html --no-optimize --public-url ./",
    "clean": "rimraf dist .parcel-cache",
    "test": "echo Error: no test specified && exit 1"
  },
  "parcelIgnore": [
    "./docs/.+"
  ],
  "author": "",
  "license": "ISC",
  "devDependencies": {
    "@types/w3c-web-usb": "^1.0.10",
    "parcel": "^2.8.3",
    "parcel-resolver-ignore": "^2.1.5",
    "rimraf": "^4.1.2",
    "typescript": "^4.9.4",
    "web-serial-polyfill": "^1.0.15"
  }
}

brianignacio5, 2025-03-13:
@Snehaaa-C  @naveenneog Could you try with PR #189 to see if issue is fixed ?

PR #189 (2025-03-18): make example scripts cross platform

## Description

Fix typescript example npm scripts to be able to platform agnostic.

Fix #169 
Fix #188

PTAL @RushikeshPatange 

## Testing

cd `examples/typescript` and run `npm run build` or `npm run dev` on both Windows and MacOS/Linux. 

It should work on any OS.

---

## Checklist

Before submitting a Pull Request, please ensure the following:

- [x] 🚨 This PR does not introduce breaking changes.
- [ ] All CI checks (GH Actions) pass.
- [ ] Documentation is updated as needed.
- [ ] Tests are updated or added as necessary.
- [ ] Code is well-commented, [... cut]
closed 2025-03-18 by brianignacio5