espressif/esptool-js #171: I don't quite understand...

0wwafa, 2024-12-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

TO WORK.

### Actual behavior (suspected bug)

I flashed the factory image for my EPS32-WROOM.
I use the esptool-js (but I get the same result with esptool.py).

After successful flashing, if I reboot it it shows:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5916
ho 0 tail 12 room 4
load:0x40078000,len:15844
load:0x40080400,len:3548
entry 0x40080600
I (29) boot: ESP-IDF v5.0.6-dirty 2nd stage bootloader
I (29) boot: compile time 04:45:59
W (29) boot: Unicore bootloader
I (32) boot: chip revision: v1.0
I (36) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  1 otadata          OTA data         01 00 00010000 00002000
I (81) boot:  2 nvs              WiFi data        01 02 00012000 0000e000
I (88) boot:  3 at_customize     unknown          40 00 00020000 000e0000
I (96) boot:  4 ota_0            OTA app          00 10 00100000 00180000
I (103) boot:  5 ota_1            OTA app          00 11 00280000 00180000
I (111) boot: End of partition table
I (115) esp_image: segment 0: paddr=00100020 vaddr=3f400020 size=1a114h (106772) map
I (162) esp_image: segment 1: paddr=0011a13c vaddr=3ff80063 size=00008h (     8) load
I (162) esp_image: segment 2: paddr=0011a14c vaddr=3ffbdb60 size=050d0h ( 20688) load
I (176) esp_image: segment 3: paddr=0011f224 vaddr=40080000 size=00df4h (  3572) load
I (178) esp_image: segment 4: paddr=00120020 vaddr=400d0020 size=129b5ch (1219420) map
I (625) esp_image: segment 5: paddr=00249b84 vaddr=40080df4 size=1ee10h (126480) load
I (677) esp_image: segment 6: paddr=0026899c vaddr=400c0000 size=00064h (   100) load
I (694) boot: Loaded app from partition at offset 0x100000
I (694) boot: Disabling RNG early entropy source...
no external 32k oscillator, disable it now.
at param mode: 1
AT cmd port:uart1 tx:17 rx:16 cts:15 rts:14 baudrate:115200
module_name: WROOM-32
max tx power=78, ret=0
v3.4.0.0
```

I used: factory_WROOM-32.bin version 3.4.0.0

Now if I try the WEB IDE, nothing happens.
It says it connects but then nothing.

Nothing using WEBUSB seems to work.

If instead I flash micropython on it, it works without any problem.

What am I doing wrong?

### Error logs or terminal output

```shell
Pasted above.
```

### Steps to reproduce the behavior

Already described above

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

CMD

brianignacio5, 2024-12-18:
Could you share the tracing output ?

brianignacio5, 2025-03-13:
I think I have found a solution for you with #190 

@0wwafa  Could you try the live demo in my fork https://brianignacio5.github.io/esptool-js/ where this is merge to test if it works for you ?