espressif/esptool-js #175: ESP32-cam flashing not working

johnyHV, 2025-01-03:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Successfully uploaded FW to the board Ai Thinker ESP32-cam with MB board. GND/R revision.
[image]

### Actual behavior (suspected bug)

Flashing FW from WEB interface https://espressif.github.io/esptool-js/ to board Ai Thinker ESP32-cam with MB board, GND/R revision don't works. FW is generate from arduino IDE. Flashing FW from Arduino IDE or oficial windows application ESP flash download tool working fine. But flashing FW from WEB app don't work.

Here is log from Arduino IDE 
```esptool.py v4.8.1
Serial port COM14
Connecting.......
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:67:24:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00197fff...
Compressed 24976 bytes to 15953...
Writing at 0x00001000... (100 %)
Wrote 24976 bytes (15953 compressed) at 0x00001000 in 0.9 seconds (effective 227.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 307.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 435.5 kbit/s)...
Hash of data verified.
Compressed 1602128 bytes to 932972...
Writing at 0x00010000... (1 %)
Writing at 0x0001b776... (3 %)
Writing at 0x0002a439... (5 %)
Writing at 0x00035464... (7 %)
Writing at 0x0004109c... (8 %)
Writing at 0x00050f01... (10 %)
Writing at 0x00059e49... (12 %)
Writing at 0x0006040f... (14 %)
Writing at 0x000685fb... (15 %)
Writing at 0x0007b10b... (17 %)
Writing at 0x00083799... (19 %)
Writing at 0x0008c024... (21 %)
Writing at 0x000949cc... (22 %)
Writing at 0x0009bada... (24 %)
Writing at 0x000a1cba... (26 %)
Writing at 0x000a83d9... (28 %)
Writing at 0x000ae4b5... (29 %)
Writing at 0x000b4326... (31 %)
Writing at 0x000b9df9... (33 %)
Writing at 0x000bf810... (35 %)
Writing at 0x000c53e1... (36 %)
Writing at 0x000cad51... (38 %)
Writing at 0x000cffda... (40 %)
Writing at 0x000d574c... (42 %)
Writing at 0x000dad9c... (43 %)
Writing at 0x000e0456... (45 %)
Writing at 0x000e5dfc... (47 %)
Writing at 0x000eb1d4... (49 %)
Writing at 0x000f05cb... (50 %)
Writing at 0x000f5dca... (52 %)
Writing at 0x000fb073... (54 %)
Writing at 0x001020b7... (56 %)
Writing at 0x001079bb... (57 %)
Writing at 0x0010cf81... (59 %)
Writing at 0x0011267a... (61 %)
Writing at 0x00117825... (63 %)
Writing at 0x0011cfbd... (64 %)
Writing at 0x00122d85... (66 %)
Writing at 0x0012867b... (68 %)
Writing at 0x0012dbbc... (70 %)
Writing at 0x00132efc... (71 %)
Writing at 0x001381ab... (73 %)
Writing at 0x0013d822... (75 %)
Writing at 0x00142ff4... (77 %)
Writing at 0x001483c2... (78 %)
Writing at 0x0014d7ec... (80 %)
Writing at 0x0015300c... (82 %)
Writing at 0x00158f55... (84 %)
Writing at 0x0015ea6d... (85 %)
Writing at 0x00165ad8... (87 %)
Writing at 0x001702f3... (89 %)
Writing at 0x001762b2... (91 %)
Writing at 0x0017b3f5... (92 %)
Writing at 0x00180bc8... (94 %)
Writing at 0x00186800... (96 %)
Writing at 0x0018bd90... (98 %)
Writing at 0x00191561... (100 %)
Wrote 1602128 bytes (932972 compressed) at 0x00010000 in 22.7 seconds (effective 564.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...```

### Error logs or terminal output

```shell
esptool.js
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting...Debug: using Classic Serial Reset
Debug: _connect_attempt default_reset
Debug: Sync connect attempt 0
Debug: Sync
Debug: Sync err Error: Invalid head of packet (0x74): Possible serial noise or corruption.
Debug: Error at sync Error: Invalid head of packet (0x74): Possible serial noise or corruption.
Debug: Sync connect attempt 1
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 2
Debug: Sync
Debug: Sync err Error: No serial data received.
Debug: Error at sync Error: No serial data received.
Debug: Sync connect attempt 3
Debug: Sync
Debug: 538052359
Debug: Connect attempt successful.

Debug: Chip Magic f01d83
Detecting chip type... ESP32
Debug: Read efuse 1073061900
Debug: Read efuse 1073061900
Debug: Read efuse 1073061908
Debug: Read efuse 1073061900
Chip is ESP32-D0WDQ6-V3 (revision 3)
Debug: Read efuse 1073061900
Debug: Read efuse 1073061900
Debug: Read efuse 1073061904
Debug: Read efuse 1073061912
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Debug: Read efuse 1073061892
Debug: Read efuse 1073061896
MAC: 08:3a:f2:67:24:94
Debug: Read efuse 1073061892
Debug: Read efuse 1073061896
Uploading stub...
Debug: mem_begin 3532 1 6144 400be000
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Debug: mem_begin 156 1 6144 3ffdeba8
Debug: check_command enter RAM download mode
Debug: check_command write to target RAM
Running stub...
Debug: check_command leave RAM download mode
Stub running...
Changing baudrate to 460800
Changed
Debug: EspLoader program
Debug: Data Length 1601968
Debug: Image Length 1601968
Debug: _update_image_flash_params keep undefined undefined
Debug: Image MD5 761fe8137e96e56708ee722eda152695
Compressed 1601968 bytes to 932891...
Debug: check_command enter compressed flash mode
Error: No serial data received.
```

### Steps to reproduce the behavior

1. Download file with FW [esp32-cam.zip](https://github.com/prusa3d/Prusa-Firmware-ESP32-Cam/releases/download/v1.1.2/esp32-cam.zip)
2. unzip file esp32-cam.zip
3. open the web page https://espressif.github.io/esptool-js/
4. click to connect on the web page, and select serial port
5. set the address to 0x10000  and file ESP32_PrusaConnectCam.ino.bin
6. click to program
[image]

### Project release version

v0.5.3

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

other (details in Additional context)

### Additional context

web browser

brianignacio5, 2025-01-06:
Could you enable tracing and share the tracing output from the `Copy Trace` button in https://espressif.github.io/esptool-js/ ?

johnyHV, 2025-01-06:
Hello @brianignacio5 . Here is output

```TRACE 219.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 221.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 223.000 Read 128 bytes: 
    7473204a756c2032 3920323031392031 | ts Jul 29 2019 1
    323a32313a34360d 0a0d0a7273743a30 | 2:21:46....rst:0
    78312028504f5745 524f4e5f52455345 | x1 (POWERON_RESE
    54292c626f6f743a 3078332028444f57 | T),boot:0x3 (DOW
    4e4c4f41445f424f 4f54285541525430 | NLOAD_BOOT(UART0
    2f55415254312f53 44494f5f5245495f | /UART1/SDIO_REI_
    52454f5f56322929 0d0a77616974696e | REO_V2))..waitin
    6720666f7220646f 776e6c6f61640d0a | g for download..
TRACE 224.000 Read invalid data: 
    7473204a756c2032 3920323031392031 | ts Jul 29 2019 1
    323a32313a34360d 0a0d0a7273743a30 | 2:21:46....rst:0
    78312028504f5745 524f4e5f52455345 | x1 (POWERON_RESE
    54292c626f6f743a 3078332028444f57 | T),boot:0x3 (DOW
    4e4c4f41445f424f 4f54285541525430 | NLOAD_BOOT(UART0
    2f55415254312f53 44494f5f5245495f | /UART1/SDIO_REI_
    52454f5f56322929 0d0a77616974696e | REO_V2))..waitin
    6720666f7220646f 776e6c6f61640d0a | g for download..
TRACE 224.000 Remaining data in serial buffer:                 
TRACE 224.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 225.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 327.000 No serial data received.
TRACE 327.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 327.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 428.000 No serial data received.
TRACE 429.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 429.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 907.000 No serial data received.
TRACE 907.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 908.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE 1010.000 No serial data received.
TRACE 4665.000 command op:0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE 4665.000 Write 46 bytes: 
[... 646 lines cut ...]
TRACE 6058.000 Received full packet: 010704003a08bb0000000000
TRACE 6059.000 command op:0x06 data len=8 wait_response=1 timeout=0.200 data=00000000ece50b40
TRACE 6059.000 Write 18 bytes: 
    c000060800000000 0000000000ece50b | ................
    40c0                              | @.
TRACE 6082.000 Read 1 bytes: c0              
TRACE 6082.000 Read 19 bytes: 
    010604003a08bb00 00000000c0c04f48 | ....:.........OH
    4149c0                            | AI.
TRACE 6082.000 Received full packet: 010604003a08bb0000000000
TRACE 6082.000 Read 6 bytes: c04f484149c0    
TRACE 6082.000 Received full packet: 4f484149        
TRACE 6083.000 command op:0x0f data len=8 wait_response=1 timeout=3.000 data=00100e0000c20100
TRACE 6083.000 Write 18 bytes: 
    c0000f0800000000 0000100e0000c201 | ................
    00c0                              | ..
TRACE 6103.000 Read 1 bytes: c0              
TRACE 6103.000 Read 11 bytes: 010f0200000000000000c0
TRACE 6103.000 Received full packet: 010f0200000000000000
TRACE 22329.000 command op:0x10 data len=16 wait_response=1 timeout=3.000 data=b0711800390000000040000000000100
TRACE 22329.000 Write 26 bytes: 
    c000101000000000 00b0711800390000 | ..........q..9..
    0000400000000001 00c0             | ..@.......
TRACE 25331.000 No serial data received.
```

brianignacio5, 2025-01-07:
I've noticed that in your output when using the Arduino IDE you flash multiple images while on esptool-js live you only flash the app image. Could you try using all 3 bin files in esptool-js with trace and see if it works ? 

I've tested both cases (only app bin file and 3 bin files) locally but I can't reproduce the issue with esp32 wrover-kit. I don't have this board since it is a 3rd party vendor and need time to get it.

Could you also test using [esptool.py](https://pypi.org/project/esptool/) with --trace (or in Arduino IDE enable tracing similar as the last you shared)  ? Since I can't reproduce the issue locally and I need to compare trace output in order to find the issue.

TD-er, 2025-01-07:
You could also generate a single bin file including the bootloader etc. using esptool.py and then try to flash that single file using the web flasher.
For some inspiration on how to generate a single file using the `merge_bin` option of esptool.py, see [here](https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/post_esp32.py) 
Or simply read the flash chip from a successfully flashed node and use that as single file to test if it is working when sending only a single file.

By reading from a successfully flashed unit you also make sure the correct sketch partition is selected, which may have been toggled after an OTA update.
Also esptool.py may change some flags based on the detected flash chip, which the webflasher may not do (not sure, just a hypothesis). Flags like DIO/QIO mode and/or flash frequency for example.

brianignacio5, 2025-01-08:
I understand @TD-er the issue is that I can't reproduce it myself locally. Either the issue is target dependent or something else I am missing which is also why I need esptool.py --trace to compare.

It does seem that esptool.py is updating flashing arguments like flash size and other stuff which esptool-js doesn't do just yet but if I try with the supplied binaries I can't reproduce the issue using esp32-wrover kit.

Jason2866, 2025-03-11:
The issue is most probably not flashing the other needed files. The have to exactly match.
A wrong not matching one ends in crash or bootloops. So the only safe way is to flash a combined factor image.

brianignacio5, 2025-03-25:
Hi @johnyHV I've got this board you used and I've tested with #191 and got it working.

This PR detects flash size from board and does spi flash command.

Could you please give it a try @Jason2866 @johnyHV  ?

Jason2866, 2025-03-25:
@brianignacio5 PR does what it is made for :-)

brianignacio5, 2025-03-26:
Yes I still need to add code for update image header and firmware SHA update. I was wondering if it should be done in the same PR or in a separated one.

The thing is, for this specific GitHub issue, seems that flashing just the single partition bin for this board (I ended up getting this board myself to fix the issue) does works just due to flash size update and spi config.

I will also add code related to SHA update and header update.

closed 2025-09-22 by brianignacio5