espressif/esptool-js #184: add get parameters in adress to define arguments for expected partitions and their addresses

jojo2massol, 2025-01-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

It would be great to be able to define in a link all partitions and addresses to simplify user experience.

Of course, the user will still have to select files, but it reduces the risk of error for newcomers.

### Use cases

eg:
`https://espressif.github.io/esptool-js?p1=bootloader.bin&p1a=0x0000&p2=firmware.bin&p2a=0x10000&p3=nvs.bin&p3a=0x9000`

Jason2866, 2025-03-13:
It is even more user friendly and avoid possible errors to merge the different binaries in one merged binary for flashing :-)

igrr, 2025-03-18:
I think this is kind of what esp-launchpad allows doing. If you are building a firmware and letting the users flash it from the browser, you can create a .toml file with information about the binary and the offset where it should be flashed. Here are some links:

- https://espressif.github.io/esp-launchpad/ (see "publish your own firmware apps")
- Example link for esp-box project, note the URL of the TOML file is provided as an argument: https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml
- Example usage in ESP-BSP project (see "Try with ESP Launchpad" column): https://github.com/espressif/esp-bsp?tab=readme-ov-file#examples

Given that ESP Launchpad was created specifically to simplify the use case of flashing apps from the browser, I think we wouldn't add this functionality in esptool-js. Esptool-js is intended to be more of a lower level component upon which higher-level apps are to be built.