espressif/esptool-js #188: examples/typescript npm run dev warnings

airjuice1, 2025-03-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

the local server will start

### Actual behavior (suspected bug)

local server does not start, warning appears

### Error logs or terminal output

```shell
[warning] ResetStrategy, defined in ./src/reset.ts, is referenced by ClassicReset but not included in the documentation.
[warning] FlashReadCallback, defined in ./src/esploader.ts, is referenced by ESPLoader.readFlash.readFlash.onPacketReceived but not included in the documentation.
[info] Documentation generated at ./docs
Ошибка в синтаксисе команды.
```

### Steps to reproduce the behavior

cd examples/typescript
npm install
npm run dev # Run local sever with example code

### Project release version

v0.5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

10

### Shell

PowerShell

RushikeshPatange, 2025-03-11:
Duplicates #169 

Hi @airjuice1 the commands used in `scripts` field of `package.json` file located at root of `example/typescript` are primarily works for a Unix-like environment (e.g., Linux or macOS) which causing issue on `Windows` machine. However, with some adjustments, they can be made to work on a Windows machine which were discussed in #169,  you can make use of workaround/solution discussed there.

But still there is a need to make `scripts` **environment agnostic**

### Listing some of the ways to achieve agnosticism in scripts execution in future:
- Use Cross-Platform npm Packages
- Use Node.js Scripts
- Use npm-run-all for Command Chaining
- Use Task Runners like gulp

brianignacio5, 2025-03-13:
@airjuice1 @RushikeshPatange  Could you try with PR #189 to see if issue is fixed ?

airjuice1, 2025-03-13:
[#169](https://github.com/espressif/esptool-js/issues/169#issuecomment-2547667036) this method helped

PR #189 (2025-03-18): make example scripts cross platform

## Description

Fix typescript example npm scripts to be able to platform agnostic.

Fix #169 
Fix #188

PTAL @RushikeshPatange 

## Testing

cd `examples/typescript` and run `npm run build` or `npm run dev` on both Windows and MacOS/Linux. 

It should work on any OS.

---

## Checklist

Before submitting a Pull Request, please ensure the following:

- [x] 🚨 This PR does not introduce breaking changes.
- [ ] All CI checks (GH Actions) pass.
- [ ] Documentation is updated as needed.
- [ ] Tests are updated or added as necessary.
- [ ] Code is well-commented, [... cut]
closed 2025-03-13 by airjuice1