espressif/esptool-js #192: Change the online demo default flash address from `0x1000` to `0x10000`

arduino12, 2025-03-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

**Hi,**

Will be very nice and will save lot of time for new users if the **[online demo](https://espressif.github.io/esptool-js/)** default flash address will be the right one for the detected MCU!

For example I use ESP32S3 and it won't boot with the default `0x1000`.
Took me a while to notice it is not `0x10000`:

[image]

The `0x10000` address seems to be the one for most ESP32 (s2 s3, c3, esp32...)
See the flash command for each **[here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flashing-firmware.html#id1)**.

Hope we can fix this soon so `0x10000` will be the default - I think it is a typo that the current value is `0x1000`.

### Actual behavior (suspected bug)

-

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

-

### Project release version

v

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Win10

### Shell

ZSH

igrr, 2025-03-16:
The base address depends on the content of the binary you are flashing as well as on the MCU.

On the ESP32, the following parts typically need to be flashed:
- bootloader at 0x1000
- partition table at 0x8000 (might differ depending on the bootloader configuration)
- application at 0x10000 (default offset, might differ depending on the partition table)

On the ESP32-S3, it is the same except the bootloader has to be flashed at 0x0 instead of 0x1000.

So there is no single "correct" offset. If you are flashing the complete binary including bootloader, partition table, and the app, you have to flash it to 0x1000 for the ESP32 or 0x0 for ESP32-S3. If your board already has the correct bootloader and the partition table, then you can flash just the app to 0x10000, or whatever is the app partition offset according to the partition table. Depending on the partition table selected, the app offset could be different.

arduino12, 2025-03-16:
**Hi,**

**Thanks for your quick reply!**

I think most of the time the bootloader and the partition table are already flashed.. so no need to re-flash them..
**Most users will only want to flash the user-app at `0x10000`- and that address is the same for all ESP32 as I understand.**

Also if the user accidentally flash to `0x1000` - it ruin the bootloader if it was previously flashed..
But if the user accidentally flash to `0x10000` - no damaged is done because the partition-table and bootloader address are well below..

So if possible please change the default address to `0x10000` - will help avoid flashing to the "wrong" memory addresses and overwriting previously bootloader and partition table!

**Arad :)**

Jason2866, 2025-03-18:
Not using a merged firmware binary has drawbacks. Are the correct files used?
Is the magic image header correct to reflect flash size, flash speed and flash mode?
By using a merged binary build with esptool.py all of that is taken care.
If the magic header is not correct this can end in flash corruption when using a Filesystem.
Been there...