espressif/esptool-js #193: Use merged_bin without overwriting NVS.

MYCAMEL222, 2025-03-30:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I want to use the merged_bin function to create a single file for the end user to upload. The problem is the partition.bin writes to 0x8000 and the factory.bin writes to 0x10000. NVS is at 0x9000. The users configuration settings are in NVS. When I write a merged bin file at 0x0, the NVS gets overwritten (I assume by padding). Is there a way to avoid that? I don't want the end user writing multiple .bin files if at all possible....Thanks for any help!

igrr, 2025-03-30:
Esptool.py has added support for "Intel hex" format, which supports sparse binaries (with gaps between regions). I think we can add support for this format to esptool-js as well. Then you will be able to generate the merged binary in ihex format and flash it without overwriting the gaps.

MYCAMEL222, 2025-03-30:
Thanks for the reply. I actually thought i was posting this on the esptool site. I have no problems using esptool-js. 

I have reposted at esptool. hopefully they have solution that i can't find by reading docs....

Thanks again.

closed 2025-03-30 by MYCAMEL222