espressif/esptool-js #194: Make flashing and monitoring easy in a single workflow.

horw, 2025-04-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Hello,

I have tried flashing and then immediately monitoring, but the current wrappers do not easily allow this. Additionally, there is a read loop running in the background that is not easy to close.

It would be nice to easily flash and monitor in a single workflow, which would allow easy switching between states.

Thank you!

### Use cases

It's convenient when you can easily flash and then monitor in the same tab without reloading it.

RushikeshPatange, 2025-04-01:
> ### Checklist
> * [x]  Checked the issue tracker for similar issues to ensure this is not a duplicate.[x]  Described the feature in detail and justified the reason for the request.[x]  Provided [...]

@horw, I think you can achieve the same functionality of **flashing and monitoring in a single workflow** by using [esp-launchpad](https://espressif.github.io/esp-launchpad/).

Steps to follow at [esp-launchpad](https://espressif.github.io/esp-launchpad/).:
- Connect your device to [esp-launchpad](https://espressif.github.io/esp-launchpad/).
- Flash your application through **Quick Start**(suitable for pre-built fw applications) or **DIY**(suitable for custom fw applications) section as per your need.
- After flash you will be redirected to **Console** section.
- Click **Reset Device** on **Console** section to monitor the device logs on terminal.

Give it a try and let us know if it works for you! If you need further assistance, feel free to reach out.

Also want to add some clarification on
> ### Use cases
> It's convenient when you can easily flash and then monitor in the same tab without reloading it.

You don't need to reload the browser tab to monitor the device that you just flashed on device
Just do following steps at  [esptool-js live example](https://espressif.github.io/esptool-js/) 
- Disconnect the device by clicking on `Disconnect` button beneath to `Program` header.
- Click on `Start` button beneath to `Console` header by selecting proper console baudrate as per fw application to monitor the device logs on terminal.

horw, 2025-04-02:
Thank you for this information! I have tried esp-launchpad, and it works very well!

However, I want something slightly different. I am currently working on JupyterLite, which has a similar form to esp-launchpad, allowing you to connect a device, flash it, and monitor it. When implementing this flow, I found that it is not so easy to do with esptool-js. Therefore, this ticket is about an implementation issue. I haven't found a simple way to do it with current classes in esptool-js ( did few work around 😄 )

tyeth, 2025-04-02:
> Thank you for this information! I have tried esp-launchpad, and it works very well!
> 
> However, I want something slightly different. I am currently working on JupyterLite, and I want to achieve a similar workflow as at esp-launchpad, allowing you to connect, flash, and monitor a device. When implementing this flow, I found that it is not so easy to do with esptool-js. Therefore, this ticket is about an implementation issue. I haven't found a simple way to do it with current classes in esptool-js ( did few work around 😄 )

Hey, please share the jupyter experiments you are doing, it would be much appreciated!

horw, 2025-04-02:
> Hey, please share the jupyter experiments you are doing, it would be much appreciated!

Yes! Sure
- https://github.com/espressif/jupyter-lite-micropython
- https://espressif.github.io/jupyter-lite-micropython/

brianignacio5, 2025-04-03:
While is not documented this is done by the https://github.com/espressif/vscode-esp-idf-web-extension to do Flash And Monitor command.

The flow typically is 

1) Create a Transport object

```ts
const transport = new Transport(port);
```

2) Do flashing

```ts
const loaderOptions = {
    transport,
    baudrate: flashBaudRate,
    terminal: loaderTerminal,
  } as LoaderOptions;
const esploader = new ESPLoader(loaderOptions);
  const chip = esploader.main();
  const flashSectionsMessage = await getFlashSectionsForCurrentWorkspace(
    workspaceFolder
  );
  const flashOptions: FlashOptions = {
    fileArray: flashSectionsMessage.sections,
    flashSize: flashSectionsMessage.flashSize,
    flashFreq: flashSectionsMessage.flashFreq,
    flashMode: flashSectionsMessage.flashMode,
    eraseAll: false,
    compress: true,
    reportProgress: (fileIndex: number, written: number, total: number) => {
      progress.report({
        message: `${flashSectionsMessage.sections[fileIndex].name} (${written}/${total})`,
      });
      outputChannel.appendLine(
        `${flashSectionsMessage.sections[fileIndex].name} (${written}/${total})`
      );
    },
    calculateMD5Hash: (image: string) =>
      MD5(enc.Latin1.parse(image)).toString(),
  } as FlashOptions;

  await chip;
  await esploader.writeFlash(flashOptions);
```

3) Reset the device and disconnect the transport object
```ts
await universalReset(transport);
  if (transport) {
    await transport.disconnect();
  }
```
4) Wait transport object to unlock (release current readLoop, which might be the issue you are facing) 

```ts
await transport.waitForUnlock(500);
```

5) Re use the same transport object and connect it again

```ts
await transport.connect(monitorBaudRate, { baudRate: monitorBaudRate });
```

6) Start a serial monitor by doing some transport rawRead and have a close function that disconnect and wait for unlock such as this

```ts
while (!this.closed) {
      const readLoop = this.transport.rawRead();
      const { value, done } = await readLoop.next();
  
      if (done || !value) {
        break;
      }
      let valStr = uInt8ArrayToString(value);
      this.writeOutput(valStr);
    }

public async close() {
    if (!this.closed) {
      this.closed = true;
      this.closeEmitter.fire(0);
    }
    if (this.transport.device.readable) {
      await this.transport.disconnect();
      await this.transport.waitForUnlock(1500);
    }
  }
```

horw, 2025-04-03:
@brianignacio5 Thank you, yes, that's actually what I am looking for! I don't know if it should be implemented on the esptool-js side, but documentation would definitely be very helpful for anyone who tries to do it. Otherwise, without going into the code, it's not easy to understand how the objects work, but that's just my opinion;

I will compare it with my code and leave a comment here as a result. Thank you!