espressif/esptool-js #198: Issue

Histywy, 2025-04-24:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

esptool.je

Serial port Webserial VendorID 0x10c4 ProductID Oxea60

Connecting... Error: Failed to execute 'open' on 'SerialPort': The port is already open.

brianignacio5, 2025-04-29:
Seems the port was not properly released before trying to start the connection.

Please provide steps to reproduce and the trace output.

brianignacio5, 2025-12-10:
Closing this issue due to lack of response. Comment here if you need more assistance.