espressif/esptool-js #206: esptool-js can't connect in Chrome 139.0.7258.66

dhalbert, 2025-08-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect https://espressif.github.io/esptool-js/ to be able to connect to an ESP32 put into ROM bootloader mode.

This works with Chrome 138.0.7204.183, which was the previous stable version of chrome on Linux (Ubuntu PPA).

### Actual behavior (suspected bug)

With Chrome 139.0.7258.66, which came out yesterday,  https://espressif.github.io/esptool-js/ no longer connects. Instead, I get the message shown in the next section.

NOTE: **This may well be a Chrome bug, not an `esptool-js` bug.** I looked at the release notes for Chrome (https://developer.chrome.com/release-notes/139) and there is not mention of any work on Web Serial. I haven't yet reported it as a Chrome bug, because I'm not familiar with what might cause the error.

I have also seen this failure on a recent macOS 26 beta, with Chrome 139.0.7258.67. I have _not_ seen the failure on Windows 11, also with Chrome 139.0.7258.67. It still connects properly.

### Error logs or terminal output

```shell
Linux (Chrome 139.0.7258.66) and macOS 26 beta (Chrome 139.0.7258.67):

esptool.js
Serial port WebSerial VendorID 0x403 ProductID 0x6001
Connecting...Error: Failed to execute 'setSignals' on 'SerialPort': Failed to set control signals.
```

### Steps to reproduce the behavior

Install Chrome 139.0.7258.66 (currently, latest) on Linux.
Go to https://espressif.github.io/esptool-js/
Connect an ESP32 via USB serial.
Press "Connect".
Choose the correct serial port in the dialog box and press OK.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Shell

Bash

### Additional context

We first noticed this today, in a tool we wrote that also uses `esptool-js`: https://github.com/adafruit/web-firmware-installer-js. I then verified it also breaks in your own https://espressif.github.io/esptool-js/.

I tracked down the Chrome versions. To repeat:
139.0.7258.66 (Linux) and 139.0.7258.6 (macOS): fails
138.0.7204.183: works
For Linux, these are the two most recent versions of `google-chrome-stable` listed here: https://www.ubuntuupdates.org/package_logs?type=ppas&vals=8

It's a lot of trouble to install a previous version of Chrome for testing. Here is a way to do it on Linux, courtesy of https://gist.github.com/amalmurali47/3715e4d28cd7f28505d0acd6cd9b8081 (slightly modified)

Two shell scripts to run:

`get-old-chrome.sh <version>`:
```sh
#!/bin/bash

export CHROME_DIR="$HOME/opt/chrome-$1"
mkdir -p "$CHROME_DIR"

wget -O /tmp/chrome.deb \
  https://dl.google.com/linux/chrome/deb/pool/main/g/google-chrome-stable/google-chrome-stable_$1_amd64.deb

sudo dpkg -x /tmp/chrome.deb "$CHROME_DIR"

sudo chown root:root "$CHROME_DIR/opt/google/chrome/chrome-sandbox"
sudo chmod 4755      "$CHROME_DIR/opt/google/chrome/chrome-sandbox"
```

`$ run-old-chrome.sh <version>`:
```sh
#!/bin/bash

export CHROME_DIR="$HOME/opt/chrome-$1"

"$CHROME_DIR/opt/google/chrome/google-chrome" \
    --user-data-dir="$HOME/.config/chrome-$1" \
    --disable-component-update
```

For example, to get the previous working version:
```sh
$ ./get-old-chrome.sh 138.0.7204.183-1
$ ./run-old-chrome.sh 138.0.7204.183-1
```

Note that you should have only one tab open that is running `esptool-js`. Otherwise a failed tab will still hold onto the `/dev/tty/USB0` or whatever even after the failure. You can see that with `lsof`:
```sh
$ lsof /dev/ttyUSB0
COMMAND   PID    USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
chrome  42094 halbert  175u   CHR  188,0      0t0 2404 /dev/ttyUSB0
```

dhalbert, 2025-08-07:
I have a suspicion the cause of this problem is this Chromium commit:
https://github.com/chromium/chromium/commit/409995d5b7f1cc865b1fc1d76afc54ffd7af83c0

> serial: Avoid combining DTR and RTS for control signals
> 
> Avoid combining DTR (Data Terminal Ready) and RTS (Request To Send) when
> setting control signals, as this does not comply with the specification.
> 
> Bug: [420689824](https://issues.chromium.org/issues/420689824)
> Change-Id: [Ib545030aaf709b432c98e6441f18d66355cba58b](https://chromium-review.googlesource.com/q/Ib545030aaf709b432c98e6441f18d66355cba58b)
> Reviewed-on: https://chromium-review.googlesource.com/c/chromium/src/+/6636770
> Commit-Queue: Jack Hsieh <chengweih@chromium.org>
> Reviewed-by: Reilly Grant <reillyg@chromium.org>
> Cr-Commit-Position: refs/heads/main@{#1472556}

tyeth, 2025-08-07:
Interesting... Thanks Dan!
That commit links here: https://issues.chromium.org/issues/420689824
Which leads to the original discussion, here:
https://github.com/WICG/serial/issues/216
and that suggests just doing multiple calls to setSignals if a different
signal order is required than the argument order [if I understand correctly]

basilhussain, 2025-08-07:
I did some digging, and I'm 99.9% sure it's a bug in the new Chrome release. *All* code that uses `setSignals` on POSIX platforms will now be broken - yours, mine, everyone's. 😤

Seems like whichever dumbass that authored the changes to the Chromium code: a) didn't read the man page for `ioctl()`, and; b) didn't test the changes at all. 🤦

The new code is passing `TIOCM_*` constants to `ioctl()` where it should instead be passing a pointer. If you open a `chrome://device-log/` tab, you'll see it logs "Bad address" errors, which would be consistent with that mistake.

franz-ms-muc, 2025-08-08:
> $ google-chrome-stable --disable-features=SerialSplitDtrAndRts
> I confirmed this works for me to regain functionality for esptool-js.

confirm!

screekworkshop, 2025-08-11:
The same problem occurs with the esp web tool, this command works for Chrome installed in a different path

`open -b com.google.Chrome --args --disable-features=SerialSplitDtrAndRts`

DreyMX, 2025-08-12:
> The same problem occurs with the esp web tool, this command works for Chrome installed in a different path
> 
> `open -b com.google.Chrome --args --disable-features=SerialSplitDtrAndRts`

Thanks!

Better than use the absolute path na look for the app.

dhalbert, 2025-08-12:
Chrome 139.0.7258.128 has turned off `SerialSplitDtrAndRts`. See https://issues.chromium.org/issues/437706191. I tested after upgrading Chrome and https://espressif.github.io/esptool-js/ works again without having to disable the feature via command line.

dhalbert, 2025-12-10:
This is fixed as of several Chrome versions ago.

I recently had a little more trouble getting Chrome to release the serial port. I think I sometimes had to restart the browser, not just close the tab. But I have not been able to reproduce this reliably.